themactep/thingino-firmware #381: WebSocket is already in CLOSING or CLOSED state.

dcjulian29, 2024-12-16:
Recently purchased 4 Cinnado D1 (t23) cameras. I've done the no-tools-install on 2 of them. One of the cameras is streaming video but the other one is not and all of the settings are not set in the streaming page of the web UI. Opening the Chrome debugger, I see a lot of "WebSocket is already in CLOSING or CLOSED state" errors on both of them. The one that has no settings set does not allow me to save any of the configuration. I'm brand new to this but am pretty tech-savy so I'm probly missing something since one of them is working and I followed the same procedure on both of them. I've rebooted the one that has no video several times and cycled the power twice. I'm running the latest version of Chrome on Windows 11 and also tried on Firefox with the same outcome.

gtxaspec, 2024-12-20:
is this still occurring in the latest build?

gtxaspec, 2025-01-01:
please re open if it's still an issue, couldn't reproduce

closed 2025-01-01 by gtxaspec

dcjulian29, 2025-01-01:
I ended up reflashing the firmware on the device that wouldn't show video or allow settings to be updated and that resolved the problem. Since then, I flashed a third one and did not have any problems with that one either so yes, unable to reproduce at this point