themactep/thingino-firmware #384: underscore in hostname renders camera offline

themactep, 2024-12-20:
Reported [here ](https://discord.com/channels/1086012062649565286/1086012062649565289/1319606241043152906)

I tested on a random camera and confirm the issue. Logread after boot:

```
Dec 20 10:08:09 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send RS (Network unreachable)
Dec 20 10:08:09 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Dec 20 10:08:10 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Dec 20 10:08:12 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Dec 20 10:08:13 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send RS (Network unreachable)
Dec 20 10:08:16 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Dec 20 10:08:17 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send RS (Network unreachable)
Dec 20 10:08:21 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send RS (Network unreachable)
Dec 20 10:08:25 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Dec 20 10:08:42 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Dec 20 10:09:18 cam_with_underscore daemon.err odhcp6c[1406]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
```

gtxaspec, 2024-12-20:
Unable to reproduce.  IPv4 and IPv6 working fine with underscore.  Those messages are not necessarily fatal, as they are generated while the routes are being sorted out. 

Did you try to ping any hosts?

gtxaspec, 2024-12-20:
[image]

themactep, 2024-12-20:
I literally have no network connection, all interfaces are down except lo.

themactep, 2024-12-20:
As soon as i change the hostname to anything without underscore and reboot, the network is back.

gtxaspec, 2024-12-20:
hmm... this must be an issue with the router or local network.  Just tried again, with a different device, working fine here.  Will need to do more research.

virmaior, 2024-12-22:
apparently the RFC makes underscores illegal in hostnames because the origin of hostnames didn't have them.
But lots of implementations have ignored this (see https://stackoverflow.com/questions/10959757/the-use-of-the-underscore-in-host-names  )

CazYokoyama, 2024-12-22:
So this not a problem of thingino. It is a problem of network such as network router, isn't it?
So close as "caused by network environment" may be the next step.

themactep, 2024-12-23:
I will add a validation to allow only the set of valid characters as per RFC.

themactep, 2024-12-24:
fixed

closed 2024-12-24 by themactep