themactep/thingino-firmware #386: sysupgrade -b u-boot upgrade not work

Andrik45719, 2024-12-21:
build firmware-2024-12-21

running
```
sysupgrade -b
```
shows usage only
```
Run as /sbin/sysupgrade -b
Usage: /sbin/sysupgrade [-x] [-f|-p|<file>|<URL>]
...
```
then
```
 sysupgrade -b -p
```
updates only u-boot without kernel and rootfs
```
Run as /sbin/sysupgrade -b -p
Creating /tmp/sysupgrade directory
Updating self
######################################################################## 100.0%
######################################################################## 100.0%
Re-running script from /tmp/sysupgrade
Run as /tmp/sysupgrade/sysupgrade -b -p
Upgrading from GitHub
Downloading firmware from:
https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-xiaomi_mjsxj03hl_t31l_jxq03p.bin
######################################################################## 100.0%
Downloading checksum from:
https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-xiaomi_mjsxj03hl_t31l_jxq03p-update.bin.sha256sum
######################################################################## 100.0%
Verifying downloaded file
thingino-xiaomi_mjsxj03hl_t31l_jxq03p-update.bin: OK
'thingino-xiaomi_mjsxj03hl_t31l_jxq03p-update.bin' -> '/tmp/sysupgrade/fw.bin'
Erasing block: 8/8 (100%)
Writing kb: 256/256 (100%)
Verifying kb: 256/256 (100%)

```

Andrik45719, 2024-12-21:
After executing 
```
sysupgrade -b -p
```
camera stop booting, no TTY serial output.
Repaired by Ingenic USB Cloner

Andrik45719, 2024-12-21:
```
sysupgrade -b -f
```

```
Upgrading from GitHub
Downloading firmware from:
https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-xiaomi_mjsxj03hl_t31l_jxq03p.bin
######################################################################## 100.0%
Downloading checksum from:
https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-xiaomi_mjsxj03hl_t31l_jxq03p.bin.sha256sum
######################################################################## 100.0%
Verifying downloaded file
thingino-xiaomi_mjsxj03hl_t31l_jxq03p.bin: OK
'thingino-xiaomi_mjsxj03hl_t31l_jxq03p.bin' -> '/tmp/sysupgrade/fw.bin'
Userland image does not have a bootloader!
Clean up temporary files.

```

tERB00, 2024-12-22:
There is no such option as "-b"
https://github.com/themactep/thingino-firmware/wiki/Updating-Firmware#from-the-camera-itself

gtxaspec, 2024-12-22:
its a new in testing option i believe

gtxaspec, 2024-12-22:
confirmed, need more testing on -b

themactep, 2024-12-22:
-p and -b is a nasty request, lol.

gtxaspec, 2024-12-22:
i think @felipecrs 's sysupgrade PR needs to be revisited

felipecrs, 2024-12-22:
I can reopen it and make any needed changes if there's any intention to merge it.

themactep, 2024-12-22:
One source of truth. We use the same firmware build for everything. No need to download third builds especially since they wont have the env.

gtxaspec, 2024-12-22:
?? whats a third build?

themactep, 2024-12-23:
We build bootloader along with the rest of the firmware, incorporating device-specific environment. You offer to download it from a different repo, built for a generic soc model, without environment. What hidden benefit did I miss here?

gtxaspec, 2024-12-23:
i crossed that out =D

themactep, 2025-01-13:
fixed.

closed 2025-01-13 by themactep