themactep/thingino-firmware #390: SSH key delete does not delete a key actually

CazYokoyama, 2024-12-26:
[image]
[image]
[image]

CazYokoyama, 2024-12-26:
The title may be incorrect. It may be "SSH key backup always detects there is a ssh key".

themactep, 2024-12-29:
please check the new version.
https://github.com/themactep/thingino-firmware/commit/09e5c5daa2b646faa467fc2793d682248f49f099

CazYokoyama, 2024-12-29:
Not solved yet on a42f6e065a58f.
[image]
[image]
[image]

themactep, 2024-12-30:
Did you compile the new firmware? Your screenshots are from the old one. Changes only committed, the will be included in tomorrow CI build.

CazYokoyama, 2024-12-30:
I uploaded total 6 screen shots. The last screen shot shows the latest git
sha, I believe. I will check later today.

CazYokoyama, 2024-12-30:
a42f6e065a58f is "cameras: reorder parameters after renaming." which is later than 09e5c5daa2, "webui: rewrite ssh host key back up form.".

themactep, 2024-12-31:
I tested, it works. Reopen if it does not for you.

CazYokoyama, 2025-01-01:
I guess the root cause is "no parameters for redirect_back".

```
root@ing-wyze-c3-2cde ~# ls -l /etc/dropbear/dropbear*
-rw-r--r--    1 root     root            83 Dec 29 06:35 /etc/dropbear/dropbear\
_ed25519_host_key
root@ing-wyze-c3-2cde ~# fw_printenv -n sshkey_ed25519
H4sIAAAAAAAAAwFTAKz/AAAAC3NzaC1lZDI1NTE5AAAAQDNw06wOEoOW1PDQeJlendrJ5imaBX3Csjy\
SCrG7gODZUSaxcKoMloZ8mhq15+q7OFfE5pDc496nxrf8XhK+h+kaY3UiUwAAAA==
# delete ssh host key backup
root@ing-wyze-c3-2cde ~# fw_printenv sshkey_ed25519
## Error: "sshkey_ed25519" not defined
root@ing-wyze-c3-2cde ~# fw_printenv -n sshkey_ed25519
root@ing-wyze-c3-2cde ~#
# back up ssh host key
root@ing-wyze-c3-2cde ~# backup=$(fw_printenv -n "sshkey_ed25519")
root@ing-wyze-c3-2cde ~# echo $backup

root@ing-wyze-c3-2cde ~#
root@ing-wyze-c3-2cde ~# if [ -z "$backup" ]; then echo aaa; fi
aaa
root@ing-wyze-c3-2cde ~# fw_printenv -n sshkey_ed25519
AAAAC3NzaC1lZDI1NTE5AAAAQDNw06wOEoOW1PDQeJlendrJ5imaBX3CsjySCrG7gODZUSaxcKoMloZ\
8mhq15+q7OFfE5pDc496nxrf8XhK+h+k=

```
The above log shows delete/backup of ssh key is working. On the other hand, redirect_back is defined as
```
# redirect_back "flash class" "flash text"
redirect_back() {
        redirect_to "${HTTP_REFERER:-/}" "$1" "$2"
}

```
However, it  is called as below in tools-ssh-keys.cgi

```
      delete)
                if [ -n "$backup" ]; then
                        fw_setenv $envname
                        redirect_back

```

CazYokoyama, 2025-01-02:
I can't re-open. I wrote my comment.

themactep, 2025-01-02:
redirect_back without parameters redirects back to the same page via GET protocol, parameters are optional and only create a message on the page after redirection. did you test it with the latest build? it does delete the key from env. what exactly does not work for you?

closed 2025-01-02 by CazYokoyama