themactep/thingino-firmware #396: webui - disable homing always on

gtxaspec, 2024-12-30:
steps to reproduce:

fresh build:
turn disable homing slider on
click save
refresh
turn disable homing slider off
click save

slider is always on
clicking DEBUG icon, webui env shows:
`disable_homing: true`
while `fw_printenv -n disable_homing` shows:
```
# fw_printenv -n disable_homing
false
```

themactep, 2024-12-30:
[image]

`get` was replaces with `fw_printenv` yesterday but somehow there is still old code in the file.
@CazYokoyama also reported about the new build which did not incude the latest changes.

themactep, 2024-12-31:
not an issue anymore with the proper code.

closed 2024-12-31 by themactep