themactep/thingino-firmware #402: webui: led buttons in preview don't work if pin is active low

sstepansky, 2025-01-01:
When the preview page is loaded, the corresponding button will be "on" or "off" depending on the state of the pin. This, however, seems to assume the pin is active high. If the pin is active low (e.g. the white led in tapo c100v5), the button will be "on", while the led is off. Clicking the button will send a `irled 0 white`, which does nothing because the led is already off (well, it just blinks once and stays off). The button very briefly turns off and then turns on again. If the led is on when the page is loaded, the effect is reversed.

Example request when the preview is loaded (white led is off):
```
{
  "time": "1735729857",
  "command": "daynight read",
  "daynight": "day",
  "color": 1,
  "ircut": 1,
  "ir850": 0,
  "white": 1
}
```
Clicking the white led button:
```
{
  "time": "1735730275",
  "command": "irled 0 white",
  "daynight": "day",
  "color": 1,
  "ircut": 1,
  "ir850": 0,
  "white": 1
}
```

This is a little bit hackish, but it could be a possible solution.

```diff
--- a/package/thingino-webui/files/var/www/x/json-imp.cgi
+++ b/package/thingino-webui/files/var/www/x/json-imp.cgi
@@ -69,7 +69,14 @@ ir940=$(irled read ir940)
 [ -z "$ir940" ] || payload="$payload,\"ir940\":$ir940"

 white=$(irled read white)
-[ -z "$white" ] || payload="$payload,\"white\":$white"
+pin=$(fw_printenv -n gpio_white)
+if [ -n "$white" ]; then
+    if [ "${pin:0-1}" = "o" ]; then
+        white=$((white - 1))
+        white=${white#-}
+    fi
+    payload="$payload,\"white\":$white"
+fi

 payload="$payload}"
```

Now the requests look like this:
Load preview, led is off, button is also off.
```
{
  "time": "1735730421",
  "command": "daynight read",
  "daynight": "day",
  "color": 1,
  "ircut": 1,
  "ir850": 0,
  "white": 0
}
```
Click white led button, led turns on, button turns on:
```
{
  "time": "1735730500",
  "command": "irled 1 white",
  "daynight": "day",
  "color": 1,
  "ircut": 1,
  "ir850": 0,
  "white": 1
}
```
I can do a PR if this solution is ok.
Thanks for such an awesome project!

themactep, 2025-01-01:
what page is that? preview?

themactep, 2025-01-01:
reading of the active state happens in `irled`, and if it returns a wrong state then it should be fixed there.

themactep, 2025-01-01:
and i think i see the issue:
```
root@ing-cinnado-d1-f8a4 ~# irled read white
gpio value 11 | tr -d '\n'
/sbin/gpio: /sbin/common: line 19: get: not found
```

themactep, 2025-01-01:
please check if this solution works for you
https://github.com/themactep/thingino-firmware/commit/5f313b96cd6bbf2482a9ce1f4f6c1a417fa1fe0d

sstepansky, 2025-01-02:
Yes, it works as expected. Thanks!

closed 2025-01-02 by themactep