themactep/thingino-firmware #420: sysupgrade seems broken - Can't upgrade as sysupgrade-stage2 has moved

numericOverflow, 2025-01-05:
Trying to upgrade my wyzev2 cam using `sysupgrade -p` from an ssh shell and am getting a 404 not found HTTP error when it tries to pull the `sysupgrade-stage2` from github

```root@webcam ~# sysupgrade -p
Run as /sbin/sysupgrade -p
Creating /tmp/sysupgrade directory
Updating self
######################################################################### 100.0%##O=#  #                                                                        
curl: (22) The requested URL returned error: 404

Cannot download https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/overlay/lower/usr/sbin/sysupgrade-stage2 to /sbin/sysupgrade-stage2
Clean up temporary files.
```
```Thingino:  wyze_c2_jxf23, master+78b6e29, 2024-11-11 02:58:50 -0500```

It looks like there were some changes/refactoring done to  sysupgrade recently.   Specifically commit 5a949a4 renamed sysupgrade-stage2, so it looks like earlier versions depending on sysupgrade-stage2 will all fail with a 404.

gtxaspec, 2025-01-05:
Stage2 isn't needed for sysupgrade.

Try
`ATTENTION! If you are trying to upgrade thingino from a really old build, you might need to update the upgrading script first. Run the following command on the camera:
`curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/overlay/lower/usr/sbin/sysupgrade && chmod +x /sbin/sysupgrade`
then run `sysupgrade -p` as usual.`

numericOverflow, 2025-01-05:
@gtxaspec Thanks.  Where was that posted? I hadn't seen it looking around, but clearly I did not look exhaustively :)

Shortly after posting I was able to make it work by modifying the script on my camera to apply the new GH_DL_URL in my local copy of `/sbin/sysupgrade`

Basically I applied commit 561650c by hand using `vi` editor.  It was only a single line to modify.  Your curl is probably safer update so I'll use that on my other cams.

Documenting it here in issues in case anyone else runs into this.

gtxaspec, 2025-01-05:
I think it's in https://github.com/themactep/thingino-firmware/wiki/Updating-Firmware

numericOverflow, 2025-01-05:
Thanks!   I updated the title of this ticked to be more error-search friendly if others encounter.

Closing this issue.

closed 2025-01-05 by numericOverflow

THEGxHOxST, 2025-02-05:
This is the corrected command, the posted one doesn't work anymore.

curl -sL -o /sbin/sysupgrade https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade && chmod +x /sbin/sysupgrade