themactep/thingino-firmware #423: "Recording in Progress" icon stuck on and Indicator LED fails to work on Wyze Cam 3 T31X RTL

lladnar00, 2025-01-07:
On a cleanly installed Wyze Cam 3 with the T31X RTL firmware from 2025-01-07, I'm observing the following issues:

1. The "Recording in Progress" icon is always flashing, regardless of whether the camera is recording.  It remains flashing whether I toggle "Enable Recording" on or off to start or stop recording, so it isn't accurately displaying the recording status.

2. The Indicator LED on the camera does not turn on when enabling recording.  I've tried both "gpio led b" and "gpio led r".  I am able to manually turn on and off the Red and Blue LEDs on the "LED/IRcut GPIO page."

Please let me know if you need any additional details.  I've tried a full upgrade of the firmware and wiping the overlay, but the issue persists. 

```
# cat /etc/os-release
NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
IMAGE_ID=wyze_c3_t31x_rtl
BUILD_ID="master+27473fd, 2025-01-06 21:39:25 -0800"
BOOTLOADER=isvp_t31_sfcnor_ddr128M
HOSTNAME=ing-wyze-c3
TIME_STAMP=1735421746
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-gab1b47ab
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.02-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.02-git"
```

lladnar00, 2025-01-08:
After some poking around, it looks like the LED indicator is expected to not work right now, although it's confusing that the setting is still available under the "Video Recording" settings of the Web UI:

https://github.com/themactep/thingino-firmware/blob/27473fdd5e745af924d3f45dcf31aeb743885297/package/prudynt-t/files/record#L92-L99

I also saw that the "Recording in Progress" icon is based on whether OpenRTSP is running, and it seems to always be running due to this:
https://github.com/themactep/thingino-firmware/blob/27473fdd5e745af924d3f45dcf31aeb743885297/package/prudynt-t/files/S96vbuffer#L40-L52

So perhaps both of the issues I noticed are expected behavior right now.

themactep, 2025-01-22:
The icon should be only reporting on recoding now, not vbuffer. Please test with the new build and confirm it fixed.

themactep, 2025-01-24:
feel free to reopen if the problem persists.

closed 2025-01-24 by themactep

lladnar00, 2025-01-30:
Yes, I'm confirming that with a new build the issue is resolved now on my cameras.