themactep/thingino-firmware #424: burn ,use SD card

ycsongx, 2025-01-08:
### this is my hardware info:
360 D706 [D816_main]
SOC: T20  WIFI: RTL8189FTV Flash: winbond 16MB Sensor: GC2033

### this is my test:
1.  burn OpenIPC by SD in u-boot，wifi works, but no driver fot gc2033,and i do not kown how to build myself
```
mw.b 0x80600000 0xff 0x1000000
fatload mmc 0:1 0x80600000 openipc-t20-ultimate-16mb-2.bin
sf probe 0; sf lock 0;
sf erase 0x0 0x1000000; sf write 0x80600000 0x0 0x1000000
reset
``` 
2. i get the thingino code, and try to build a gc2033.ko for T20，i do, but it can't work with openipc
it seems something wrong with tx-isp-t20.ko and  sensor_gc2033_t20.ko [symbloy  ot param ]
3. i burn thingino by SD in u-boot,  use openipc code [above] , just chage the bin file[ thingino-xiaomi_xiaofang_t20l_jxf22.bin ], it works, but it can't store configs. when i reboot, it will works like a new one, etc: rootpasswd,  TZ , hostname, wlanssid, wlanpass...... 
4. i read wiki and chabge something, it still can't store configs
```
watchdog 0;
setenv baseaddr 0x82000000;
.........
```
5. i change a thingino bin,  thingino-wyze_c2_jxf22.bin, then, the u-boot seems can't work
[ i do not think this is a bug , or it because of me, i exec "reset" and "remove the SD card quickly"]
```
Thingino U-Boot for Ingenic T20L SPL 2013.07 (Dec 28 2024 - 21:35:46 UTC)
pll_init:356
l2cache_clk = 375000000
pll_cfg.pdiv = 8, pll_cfg.h2div = 4, pll_cfg.h0div = 4, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf = 30, nr = 1, od0 = 1, od1 = 1
cppcr is 03c05100
CPM_CPAPCR 03b0890d
nf = 42, nr = 1, od0 = 1, od1 = 1
cppcr is 02a04900
CPM_CPMPCR 07d0c90d
nf = 50, nr = 1, od0 = 1, od1 = 1
cppcr is 03204900
CPM_CPVPCR 0320490d
cppcr 0x9a794410
apll_freq = 712704000
mpll_freq = 1000000000
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 500000000
cclk  712704000
l2clk 356352000
h0clk 250000000
h2clk 250000000
pclk  125000000
DDRC_DLP:0000f003
```
6.  i try to boot from SD card use  "u-boot-isvp_t20_msc0.bin", cant's work ,it seems like need gpio config.
7.  i will try to use CH341 to burn a new binfile 

### Is anyone willing to give me some advice:
1. why i burn the thingino bin, it cant't store configs
2. how can i build a gc2033.ko work with openipc or thingino

thanks

gtxaspec, 2025-01-08:
Hi, do you have photos of the PCB for your unit? we can try to make it work.

gtxaspec, 2025-01-11:
@ycsongx any updates?

ycsongx, 2025-01-12:
sorry. i am an IT operater, not a coder.
i get this hardware  from my old IPC, [ i just try my best to make it work, not  must], no any develop info,  
i need to buy somwthing to use the ch341a, it may needs some time
i try to built a gc2033.ko this two days, and i will try openipc, buecause i think i like to user a .bin file to the most hardware.
for other functions, i will read the source and try to make it works.

i have no logs, but you can some info from my test  using openipc:
 https://github.com/OpenIPC/firmware/issues/1656

thanks for all your help.

gtxaspec, 2025-01-12:
@ycsongx do you have a photo of the PCB board inside? we need to determine the CPU model.