themactep/thingino-firmware #425: Feature Request: Record motion only

ajbastien, 2025-01-08:
Instead of recording 24/7. It wold be great if the Motion Guard had an option to trigger recording for x seconds

themactep, 2025-01-16:
This is how sending motion notifications to telegram works now. If it proves to be mature enough, i will extend it to other means.

ajbastien, 2025-01-16:
I tried FTP only. But FTP and telegram only send a picture. I’d like a video for FTP.

beginner3456789, 2025-04-05:
In /etc/thingino.config change ftp_send_video="true" to ftp_sendvideo="true" and reboot or restart prudynt. Look in /tmp for the vbuffer.mp4 to verify prudynt is ready to respond to a motion.

I only get a 10 second record of a motion event except would prefer a continuing record of the motion event as long as that lasts. I don't save any continuous videos or timelapse snapshots to the mmcblk0p1 and just get the videos sent to ftp as I wanted. I wonder why the motion video isn't saved to the mmcblk0p1 card as an option.

move-some, 2025-04-17:
I would love to see this feature, too.

giggo, 2025-08-17:
Is "ftp_send_video="true"" still working in the latest versions?