themactep/thingino-firmware #426: Issue with MAC address during setup

miko007, 2025-01-08:
I flashed a Cinnado cam with `thingino` using the SD card method. I then connected to the AP and filled in the credentials for my wifi in the setup wizard. After that, the cam rebooted and the AP was gone, but i could not find it in my local network. I reset the cam like 50 times, it never could be found in my network.

After playing around for a few hours, i noticed, that the wizard printed out a MAC address for the camera. I then searched for that MAC address in the leases of my wifi router and lo and behold: **that MAC address was the exact MAC address of the laptop i was setting the camera up from**. The router then obviously got confused and never assigned an IP address to the device.

I then had to set it up in AP mode, connect via SSH and set some random other MAC address via `fw_setenv wlanmac xx:xx:xx:xx:xx:xx`. After that, everything worked as expected.

I do not know, if this was a really bad coincidence, or if the wizard somehow grabbed the mac of my laptop and used that, but nevertheless i think we should somehow provide at least documentation on how to resolve this issue, when it occurs.

nicolasroche898, 2025-01-10:
The probability of this happening is 2**-46 = 1 in 70368744177664. There is probably something else at play.

What is the lucky MAC address?

miko007, 2025-01-10:
I am using an M2 MacBook. `macOS` does spoof the MAC address for every wifi one connects to, for security reasons. The specific MAC address was `3A:A9:90:08:BE:11` which i also checked. It is the actual address, macOS uses for this specific wifi. It was also the MAC address, the thingino wizard picked during setup. There was no confusion or anything, both devices  are comfirmed to have used the exact same MAC address.

I am certainly aware of the odds of this happening *"by accident"*, thats why i raised this issue. I do not know the inner workings of thingino, which is why i suspected the wizard grabbing the MAC address of my client and using that for some odd reason. If this is not the case, then i certainly should play the lottery.

nicolasroche898, 2025-01-10:
All thingino devices mac address start with "02". Yours doesn't. I googled your mac address and didn't get any results, so you're the first. The first suspect is still your router/AP, typical scenario is some DHCP address exhaustion or bad assignement by the router, combined with some bad router logic possibly in the GUI layer (mapping from non-unique ip address to mac address, instead of unique mac address to ip address).

If it happens to someone else, we can revisit the issue.

ref about "02" mac address:

https://github.com/gtxaspec/ingenic-u-boot-xburst1/blob/9b99f7706993a75623d33737aa89c3944b3f6d70/arch/mips/lib/board.c#L118

miko007, 2025-01-11:
to be fair, there is nothing wrong with the router. I confirmed on both devices that they actually had the same MAC address using `ifconfig`. also `fw_printenv` showed `wlanmac=3a:a9:90:08:be:11`, so i am pretty sure, this really happened *(although, still not really believable)*.

I have no idea, how this MAC got rolled by thingino, but it did.

The main takeaway for me probably is, that i was able to fix this myself due to my technical understanding and decades long experience with networks, linux systems and code. For any random person to stumble on something similar would probably drive them insane. Documentation about the `fw_printenv` and `fw_setenv` *(or any special built-in commands on thingino)* is pretty sparse and scattered all over the wiki; I think this could be improved on heavily. A central page about the architecture of the thingino OS could be very helpful, even for experienced people. I spent like 20 minutes to stitch together all the network layer used *(wpa_supplicant, ifupdown etc.)* and i am still not sure which init system is used *(i also would recommend putting a symlink from `/usr/bin/vim` to `/usr/bin/vi` to not confuse the hell out of younger people ^^)*.

I would volunteer on that, but i feel there are people here, that would have a much easier time putting all that information together, as they already know it, and i would have to fumble around for probably weeks.

gtxaspec, 2025-01-11:
Very unusual for this MAC to show up, especially since there is logic in place to generate the mac according to certain parameters.  Regardless, if this happens again, please let us know. 

and FYI, the wiki is open for all to edit, provided you have a github account.  The more eyes and hands on it, the better it will be =D

unless you know of a way to grow more hands... :laughing:

gtxaspec, 2025-03-03:
please try the latest builds and report if this is still ocurring.

miko007, 2025-03-04:
> please try the latest builds and report if this is still ocurring.

i think this was a once in a lifetime occurence. i do unfortunately have no ability to reproduce this. the MAC never got rolled again.

i went in during AP mode, changed the MAC manually and the problem was solved. i since flashed multiple cams the same way, this did not occur once after.

closed 2025-03-04 by gtxaspec