themactep/thingino-firmware #428: WyzeV3 prudynt stream failing

Responseless, 2025-01-09:
Multiple WyzeV3's (t31zx and t31x models) video stops and gain goes to 0 at random times. Video does not restart without a reboot. I use almost default settings with the substream disabled. With Substream enabled the crashes are more prevalent. I use AAC audio, but OPUS causes crashes too. Using go2RTC 1.9.8. I've tried tweaking the config over the last 2 weeks but nothing seems to fix the issue. 

Diagnostics: https://termbin.com/lmcr

[20948.446487] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[20948.446487] 00000000 
[20948.446504] epc = 77597bd0 in libc.so[7752c000+b8000]
[20948.446528] ra  = 774df634 in libimp.so[7741c000+107000]
[20948.446538] 
[20948.507228] gc2053 stream off
[20948.679520] codec_codec_ctl: set CODEC_TURN_OFF...

scopedial, 2025-01-10:
I am seeing similar behaviour 
https://termbin.com/dtat

plat2on1, 2025-01-12:
I'll third this, I thought it was a hardware issue as only one of my cameras out of a 2-pack does it. but I guess not.

Responseless, 2025-01-14:
I've updated a t31zx (Partial upgrade in the web also gets stuck on stopping the vbuffer on all my v3's. I needed to ssh and run ' /etc/init.d/S96vbuffer ' for it to continue)

- Changed RTSP Password
- Disabled Thingino logo on main stream
- Disabled Substream
- Changed audio to AAC
- Hit Save
- Reboot

Seems ok so far but i will let you know if it's better or not after watching logs today.

plat2on1, 2025-01-14:
It's a no go for me.

BUILD_ID="master+80cf51a, 2025-01-12 21:34:39 -0800"

[  233.392681] do_page_fault() #2: sending SIGSEGV to prudynt for invalid read access from
[  233.392681] 00000000 
[  233.392710] epc = 77159bd0 in libc.so[770ee000+b8000]
[  233.392750] ra  = 770a1634 in libimp.so[76fde000+107000]

gtxaspec, 2025-01-14:
please share your prudynt config, or even better, the share the logs from `thingino-diag`

plat2on1, 2025-01-14:
https://termbin.com/8u57

It's odd, two cams from the same box one has 20 days uptime the other rarely goes over 12 hours without crashing.

gtxaspec, 2025-01-14:
are you using the wyze supplied power supply? they don't seem to last long, and start to drop voltage.  once the voltage stops dropping, weird stuff starts happening.  try swapping the power supply and see if that fixes it.

plat2on1, 2025-01-14:
First thing I tried was swapping the power supplies.

gtxaspec, 2025-01-14:
try:
edit `/etc/modules.d/avpu`.  change the clk to `400000000`
see if that makes a difference

plat2on1, 2025-01-14:
That was my thought above as well, I built the firmware with the config set 400MHZ but it didn't help. I'll try with this newer build though.

Responseless, 2025-01-15:
Update after upgrading yesterday: no real change, still common crashes. Will try swapping the power adapters around to see if it makes a difference. Upgraded all wyze v3's again to todays release.

plat2on1, 2025-01-15:
No change here either with the new build and 400MHZ AVPU. How many cameras do you have that are crashing vs working?

gtxaspec, 2025-01-15:
are we seeing these crashes with AAC? anyone running PCM or OPUS? Or with audio disabled?

Responseless, 2025-01-15:
> No change here either with the new build and 400MHZ AVPU. How many cameras do you have that are crashing vs working?

I have 4 v3's and all of them crash but at different times without any sort of pattern. 2 of them more often than others. I am monitoring them with HA and rebooting them with ssh automatically if the fps drops to 0 for > 1 min.

I still need to swap the power adapters around to see if that changes anything when I get home tonight.

Responseless, 2025-01-16:
The Wyze v3 t31x's seem more stable without changing anything other than software.
The Wyze v3 t31zx's are the issue now. I updated them both to todays latest firmware and they are still randomly freezing. I did change the power adapter on one of them and that one is more stable -- for now. It's a waiting game to see when they freeze. Will change the other ones power supply tonight.

themactep, 2025-01-16:
Run streamer in console in foreground, watch for errors when it freezes
```
killall prudynt
prudynt
```

Responseless, 2025-01-20:
Updated the power supplies and didn't make a difference, multiple freezes daily. t31zx are causing the most problems. Upgraded to latest 01-19 today. Will continue testing.

Responseless, 2025-01-22:
Latest versions 2025-01-19 onwards seem a lot more stable for the t31zx, gc2053. All my cameras have been up 15 hours +. I wont upgrade them for a while to see how long they can run. Thanks for your work @themactep

themactep, 2025-01-22:
A round of applauds goes to @gtxaspec and @Lu-Fi , they are the brains behind these changes,

Responseless, 2025-01-23:
Cheers @gtxaspec and @Lu-Fi

closed 2025-01-23 by Responseless