themactep/thingino-firmware #429: GNCC C2 Installation

yeyeto2788, 2025-01-09:
Hey guys! 

I want to install the firmware into the GNCC C2 which has the following configuration (T23N,_SC2336,_ATBM6012BX, 8MB) and seems like the Galayou G7 also has a similar configuration (T23N, SC2336, ATBM6012BX, 8MB) notice only the difference on the underscore on the bluetooth/wifi chip and the image sensor. 

Should I follow the same installation guide for the Galayou G7 but changing the bin file used for it?

I can also provide pictures of the internals of the camera if needed.

Cheers.

themactep, 2025-01-10:
is it this camera? https://uk.gncchome.com/products/c2-indoor-security-camera-1080p
it's supported. 
[image]

yeyeto2788, 2025-01-10:
Yes, that's the one. But I was more interested to know what guide to follow. I ended up flashing it directly onto the flash using the CH341A but maybe since the Galayou G7 seems to have the same specs maybe we can state in the docs that in order to flash the GNCC C2 you can follow the guide from Galayou.

I have taken the opportunity to dump the original firmware out of 4 cameras and also take some pictures of the internals if needed.

Cheers.

P.D: I couldn't wait for a reply but if it is needed to write original firmware back into the camera to try another flashing method ping me.