themactep/thingino-firmware #433: recording doesn't start upon boot/restart with large SD card

nicolasroche898, 2025-01-11:
symptom:

recording doesn't start upon boot/restart.
it can be manually started later on.
SD card is a 100% healthy high endurance card specified for continuous loop video recording with more than 99% lifetime remaining (in terms of FDW full disk writes).
not an sd card problem.

root cause:

the exfat mount operation, although it returns quickly, takes >10seconds until files are available in mount point
during this time, the S96record script already has run, failed to detect the recording directory, and quit...

fix:

add this to the beginning of S96record script:

```
# wait for sd card mount to finish
if [ "$(awk '{print int($1)}' /proc/uptime)" -lt 20 ]; then
  sleep 10
fi
```

themactep, 2025-01-16:
Is it still an issue with the last builds?

themactep, 2025-01-24:
feel free to reopen if the problem persists.

closed 2025-01-24 by themactep