themactep/thingino-firmware #436: Cinnado D1 - prudynt  possible memory leak

ikout, 2025-01-12:
A day or two after upgrading the camera it stopped streaming and at the logs I see a crash and the message:
"Out of memory: Kill process 1784 (prudynt) score 517 or sacrifice child"

Logs are at: https://termbin.com/7xz7

I'm going to try the latest release (Jan 11th) and see if that changes...

gtxaspec, 2025-01-13:
please try the latest build, there has been updates to the streamer as of todays build.

ikout, 2025-01-13:
Sure I'll give it a go, I had some power outages so I can't say for sure that there was any changes with the Jan 11th version.
But I saw your comment on the #438 so I'll wait for the next version to get that fixed as well.

gtxaspec, 2025-01-13:
the current posted build has #438 fixes included.

ikout, 2025-01-13:
Good, I'll give it a go then.

ikout, 2025-01-16:
The video stream stopped again but if I read the logs correct there was no crash mentioned at the dmesg output, only some wifi disconnect and the following message:

[185539.869017] do_page_fault() #2: sending **SIGSEGV** to Encoder-0 for invalid read access from

But at the logcat I see the following message several times:
IMP_Encoder_GetPool(55):chnNum: 0 not bind pool, If you don't use MemPool, don't pay attention!!

Camera Uptime:️ 2d 16h 53m
Logs:  https://termbin.com/wa90

ikout, 2025-01-16:
One follow up, I don't think the "restart streamer" button works.

In my case like above, I don't think that anything else than rebooting the camera will recover it, but I've tried it at some other occasions before and I've not seen the PID to change or anything to be happening when I press it.

themactep, 2025-01-16:
`service restart prudynt` in console will restart the streamer process. The button in webui is for restarting separate threads which does not help if the core process has crashed.

ikout, 2025-01-16:
"service restart prudynt in console"
Good to know! 

Do you think that I should upgrade to the latest version? Or it will be best that you look at the logs first in case there is something that stands out?

themactep, 2025-01-24:
is it still an issue?

ikout, 2025-01-24:
Yes..... :(
Maybe not a "Out of memory" scenario now but it crashes.... if you want to have this as a separate issue let me know.

Release: master+184692a, 2025-01-21 21:51:59 -0800
Uptime:️ 0d 21h 18m 37s

dmesg:
[49815.401689] do_page_fault() #2: sending SIGSEGV to Encoder-0 for invalid read access from
[49815.401689] 00000000
[49815.401707] epc = 77da1b1c in libimp.so[77d5b000+ef000]
[49815.401732] ra  = 77da1b10 in libimp.so[77d5b000+ef000]
[49815.401745]
[49815.420925] sc2336 stream off
[49815.604092] wait stop  q->num_buffers=2 q->done_count=0
[49816.004058] streamoff stop！！
[49816.024073] wait stop  q->num_buffers=2 q->done_count=0
[49816.424059] streamoff stop！！

ikout, 2025-01-26:
New crash with 184692a
Uptime:️ 2d 23h 16m 10s

The only things that I change from the defaults are setting FPS to 20 and some OSD changes...

[145611.073811] do_page_fault() #2: sending SIGSEGV to S96vbuffer for invalid write access to
[145611.073811] 7fd83070
[145611.073829] epc = 7771d33c inra  = 7771d470 in
[145611.073850] BUG: sleeping function called from invalid context at include/linux/freezer.h:56
[145611.073861] in_atomic(): 1, irqs_disabled(): 0, pid: 2033, name: S96vbuffer
[145611.073873] CPU: 0 PID: 2033 Comm: S96vbuffer Tainted: G           O 3.10.14__isvp_pike_1.0__ #1
[145611.073881] Stack : 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[145611.073881]           00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[145611.073881]           00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[145611.073881]           00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[145611.073881]           00000000 00000000 00000000 00000000 00000000 00000000 00000000 806c3d50
[145611.073881]           ...
[145611.073967] Call Trace:[<8001f884>] 0x8001f884
[145611.073984] [<8001f884>] 0x8001f884
[145611.073992] [<800484e0>] 0x800484e0
[145611.074001] [<800476b8>] 0x800476b8
[145611.074009] [<8001e428>] 0x8001e428
[145611.074018] [<800629f0>] 0x800629f0
[145611.074027] [<80035718>] 0x80035718
[145611.074035] [<80049110>] 0x80049110
[145611.074044] [<8001f0bc>] 0x8001f0bc
[145611.074052] [<8001b500>] 0x8001b500
[145611.074063]
[145611.074079] note: S96vbuffer[2033] exited with preempt_count 255

ikout, 2025-02-18:
I've not seen this for a while, I have a camera with the 2025-02-03 release and it's up for 12 days and not seen any crash so far...

So for now, unless the newer releases break something again I call it "fixed".