themactep/thingino-firmware #441: [send2telegram sendphoto] magic value doesn't match

mpmc, 2025-01-13:
Hi, awesome firmware! 

Currently testing it on 2 Aoqee C1 cameras and spent the last few hours trying to work out why I wasn't receiving photos via telegram :(. 

The magic value used [here](https://github.com/themactep/thingino-firmware/blob/80cf51a0edc81cf28c0077806c9ba8b149bed747/overlay/lower/usr/sbin/send2telegram#L90) isn't valid for the C1  so the script fails over to sending a it as document instead which only accepts PDF & ZIP files.

Changing this to `ffd8ffdb004300030202` and it works!

themactep, 2025-01-14:
I changed the magic to more permissive, please test and report of it fixed the issue.

mpmc, 2025-01-15:
Awesome. Thank you!