themactep/thingino-firmware #443: Wuuk Y0510 eventually crashes

zoof, 2025-01-14:
I have the Wuuk Y0510 SC4336p and I have found that after about 12-16 hours, it crashes and is unreachable.  So far, I have tried 3 different firmware images.  I'm happy to provide any diagnostic information to help debug this.  The current firmware version is wuuk_y0510_sc4336p, master+75d2d41, 2025-01-12 01:11:34 -0800.

wltechblog, 2025-01-22:
when the device is unreachable, does the chime play when tapping the button?

zoof, 2025-01-22:
I'm not sure why it kept crashing but it's been up and accessible for 6+ days now.  I'll close the issue and reopen it if it crops up again.  Thanks for your response.

themactep, 2025-01-24:
feel free to reopen with details if you have new crashes.

closed 2025-01-24 by themactep