themactep/thingino-firmware #447: Video Recording uses a "dummy" hostname when enabled & Limit behavior change?

ikout, 2025-01-15:
After upgrading (master+80cf51a) I went to re-enable the video recoding and I noticed a few things....

1. The default format of the recording path has changed.
2. The hostname is added at the path BUT when enabled it adds a "dummy" hostname, in my case it was "Janostname"
3. The limit at the webui at least it says now "usage in GiB", before it was "%" if I'm not wrong, so do we define now the space we want the video files to take? i.e. if I add "10" it means 10GB on the SD no matter if the SD is 16/32/256GB?

Main "issues" are the default format and the "dummy" hostname, once you save the settings I believe it's using the correct one but if you just enabled it, uses that dummy one.
Also let's clarify the "new" limit behavior, and I also noticed that there is no "Video Recording" section at the Wiki, I believe it was one about 2-3 weeks ago.

Final topic that I could open it as a separate issue, if you only enable timelapse, there is no "monitor/limit" for the disk space and it fills the SD card....

themactep, 2025-01-16:
Do not squash multiple issues into one, these are hard to track and close. Open separate tickets next time.

Default path is a mere convenience. If can be not set at all leaving it up to you to use whatever format you like. 

Hostname parsing as you described it seems broken as it parses date before hostname. Will be fixed.

Limit value is optional. Do not put any if you want to use the whole card.

ikout, 2025-01-16:
I know it was not ideal, to add all these at once... But I was not sure which one is related to the other since are all at the same "settings" page.

In any case let's keep this for the hostname behavior and if possible clarify the "limit" value, if I put "10" does that mean 10% or 10GB?
And if I don't put anything as you said since that's optional, will not that fill up the SD and cause issues?

themactep, 2025-01-16:
Script will always rotate records when reaching the limit of capacity, either physical defined by the size of the media or virtual set in the form. If you set the limit to say 10GB then even if you record on a network share the camera will only use 10GB of its storage space and will start deleting older records to stay within limits, so you can set up multiple camera to use the same share without fear that they overflow the space.

themactep, 2025-01-16:
@ikout how old is your build? i cannot respoduce that. `record` correctly creates files with hostname path: `/mnt/mmcblk0p1/ing-cinnado-d1-f8a4/2025/01/16/22/42-46.mov`

ikout, 2025-01-16:
As I mentioned at the first post this happened when I upgraded at the "80cf51a", I have upgraded now to the "bab79af" now (Due to the #436).
Since after every upgrade the timelapse settings are reverted/disabled I re-enable it and noticed this behavior.  With the 80cf51a (or one-two versions back) I believe there were the changes to the webui with the limit from the old "85%" to this new one that's "usage in GiB".

I think at before the upgrade to 80cf51a I downgraded to an older release (with downloading an old version locally and used the "Flash Operations" & full upgrade) because something was not working well and after a couple of days I went to the 80cf51a. 
I don't know if that contributed to this behavior.

UPDATE: While I was writting the above I recalled that with the last upgrade I have not re-enabled the Video Recording and I just did that and same issue happened:

```
root@Cinnado-01 ~# date ;ls -lh /mnt/mmcblk0p1/
Fri Jan 17 01:03:50 EET 2025
drwxrwxrwx    2 root     root      192.0K Jan 10 23:59 20250110
drwxrwxrwx    2 root     root      192.0K Jan 11 23:59 20250111
drwxrwxrwx    2 root     root      192.0K Jan 12 23:59 20250112
drwxrwxrwx    2 root     root      160.0K Jan 13 18:16 20250113
drwxrwxrwx    2 root     root       64.0K Jan 14 23:59 20250114
drwxrwxrwx    2 root     root      192.0K Jan 15 23:59 20250115
drwxrwxrwx    2 root     root      160.0K Jan 16 17:28 20250116
drwxrwxrwx    6 root     root       32.0K Jan  8 10:14 Cinnado-01
drwxrwxrwx    4 root     root       32.0K Jan 17 00:00 timelapse
root@Cinnado-01 ~#
root@Cinnado-01 ~# date ;ls -lh /mnt/mmcblk0p1/
Fri Jan 17 01:04:24 EET 2025
drwxrwxrwx    2 root     root      192.0K Jan 10 23:59 20250110
drwxrwxrwx    2 root     root      192.0K Jan 11 23:59 20250111
drwxrwxrwx    2 root     root      192.0K Jan 12 23:59 20250112
drwxrwxrwx    2 root     root      160.0K Jan 13 18:16 20250113
drwxrwxrwx    2 root     root       64.0K Jan 14 23:59 20250114
drwxrwxrwx    2 root     root      192.0K Jan 15 23:59 20250115
drwxrwxrwx    2 root     root      160.0K Jan 16 17:28 20250116
drwxrwxrwx    6 root     root       32.0K Jan  8 10:14 Cinnado-01
drwxrwxrwx    3 root     root       32.0K Jan 17 01:04 Janostname    <<== After re-enabling the Video recording...
drwxrwxrwx    4 root     root       32.0K Jan 17 00:00 timelapse
root@Cinnado-01 ~#
```

If it helps I got the logs now, just I re-enabled it and has this settings: https://termbin.com/zx4n

I also see at the "/tmp/webui.log" that it "reports" the correct path but at the SD I see that "Janostname"

```
2025-01-17 01:03:00 [4724:timelapse] Plugin timelapse initialized
2025-01-17 01:04:00 [5901:timelapse] Plugin timelapse initialized
2025-01-17 01:04:12 [6171:] 
stream_number: 0
stream_endpoint: ch0
stream_fps: 25
stream_height: 1080
stream_width: 1920

2025-01-17 01:04:13 [6171:] Making directory /mnt/mmcblk0p1/**Cinnado-01**/2025/01/17/01
2025-01-17 01:04:13 [6171:] Space available: 9566400 KiB
2025-01-17 01:05:00 [6804:timelapse] Plugin timelapse initialized
2025-01-17 01:06:00 [7297:timelapse] Plugin timelapse initialized
```

themactep, 2025-01-17:
Try a new build tomorrow or pull changes and compile before that. I pushed full rewrite of both time lapse and video recording. No more stray directories and badly parsed hostnames.

themactep, 2025-01-17:
Janostname is `%hostname` which got to date parser earlier than to hostname parser. `%h` became recent abbreviated month name - Jan.

themactep, 2025-01-22:
Please confirm the fix works for you.

ikout, 2025-01-22:
I believe it's fixed since f2cd8ca, I'm going to upgrade to the latest as well to be sure, but let's close it for now.

closed 2025-01-22 by ikout