themactep/thingino-firmware #449: Timelapse has no disk space limits, it will fill up the SD card.

ikout, 2025-01-16:
If you only enable timelapse, and not have Video Recording enabled, there is no "monitor/limit" for the disk space and it fills the SD card....

Can this be added?

And potentially it will need to be either aligned with the limit of the Video Recording or have separate values for each.
i.e.
Timelapse: 5GB
Video Recording: 10 GB

themactep, 2025-01-16:
timelaps implies that you retain all frames before you combine them into a timelapse clip. having a timelapse that deletes older files sounds silly. use a larger card or a network share.

ikout, 2025-01-16:
Even if I use a network share I don't want it to take over all the share space, it would be best that there is an option to say keep the last 5 days and delete them automatically if no action is made at them. And since the firmware itself is actually building the timelapse but expects a 3rd party to do that, it should protect itself from running out or resources.

BTW, I can't find at the wiki how it's supported by Thingino, I've seen the U-Boot section but it's mentioning "how to boot a device using U-Boot over NFS".

What kind/type of network shares are supported by Thingino?

gtxaspec, 2025-01-16:
NFS is supported out of the box.  You can add CIFS if you compile your own build.

themactep, 2025-01-16:
i like the idea to limit the depth in full days. i will add that.

ikout, 2025-01-16:
> i like the idea to limit the depth in full days. i will add that.

That will be great!

themactep, 2025-01-22:
Test in the new build.

ikout, 2025-01-22:
I'll upgrade to the latest build and get back to you in 1-2 days, I have about 10 days worth of folders so let's see.
At the moment with the f2cd8ca the old folders are not deleted.

I'll test the new one and let you know.

themactep, 2025-01-23:
does it still create convert files? i thought i removed that, lol.

themactep, 2025-01-29:
After a week of consideration I conclude. Running recorder on the camera is a stupid enough solution especially after we've made RTSP available on these cameras. Making the camera responsible for scanning remote, way more powerful machine, sounds even worse than that. I can only suggest to create an external garbage collector on the share side and run it periodically to prune the storage.