themactep/thingino-firmware #452: webui: timelapse refactor, missing .jpg at the filenames

ikout, 2025-01-18:
From a quick test with new build f2cd8ca, nice on the re-layout, but I noticed one small issue, at the saved files the extension is missing:

```
nfs-server:~# ls -lh /mnt/nfs-share/Cinnado-01/timelapses/20250118/
total 5M
-rw-r--r--    1 root     root      752.0K Jan 18 12:15 20250118T121507
-rw-r--r--    1 root     root      752.1K Jan 18 12:16 20250118T121608
-rw-r--r--    1 root     root      731.0K Jan 18 12:22 20250118T142204
-rw-r--r--    1 root     root      735.2K Jan 18 12:23 20250118T142313
-rw-r--r--    1 root     root      762.5K Jan 18 12:24 20250118T142401
-rw-r--r--    1 root     root      762.1K Jan 18 12:25 20250118T142500
-rw-r--r--    1 root     root      730.6K Jan 18 12:26 20250118T142607
nfs-server:~#
```

themactep, 2025-01-18:
do you have `.jpg` in the filename template field?
[image]

themactep, 2025-01-18:
ah, i see. it's missing from defaults. fixed.

themactep, 2025-01-22:
Is it good now?

ikout, 2025-01-22:
Tested the latest version 184692a, this is fixed now.

closed 2025-01-22 by ikout