themactep/thingino-firmware #453: Web UI can't render correctly

unijiang, 2025-01-19:
I have updated the xiaomi_mjsxj03hl_t31n_jxq03p  to the latest firmware (master+486af9d, 2025-01-18 23:32:16 -0800). The web ui can't render correctly.
[image]

themactep, 2025-01-22:
Static ip?

unijiang, 2025-01-22:
I bind the IP/MAC in OpenWrt router.

unijiang, 2025-01-22:
> Static ip?

I bind the IP/MAC in OpenWrt router.

themactep, 2025-01-22:
Check /etc/resolv.conf for content. If the file is empty, add `nameserver 8.8.8.8` to it

unijiang, 2025-01-22:
> Check /etc/resolv.conf for content. If the file is empty, add `nameserver 8.8.8.8` to it

The resolv.conf:
```
search lan
nameserver 8.8.8.8 # new added
nameserver 192.168.2.1
nameserver fd1e:bce5:916d::1 # added by odhcp6c
```

The Web UI still corrupt.

themactep, 2025-01-22:
Do you have access from your computer to internet? Can you open thin link? https://cdn.jsdelivr.net/npm/bootstrap@5.3.3/dist/css/bootstrap.min.css

unijiang, 2025-01-22:
> Do you have access from your computer to internet? Can you open this link? https://cdn.jsdelivr.net/npm/bootstrap@5.3.3/dist/css/bootstrap.min.css

Thanks, that's the root cause, a service in my OpenWrt blocks this CDN domain, I did some configuration, the Web UI comes back now. Was this CDN added recently? This issue just happens after I updated the firmware.

closed 2025-01-22 by unijiang

themactep, 2025-01-22:
Yes, it is a recent change, maybe a week or less. You can set the paranoid mode in web ui settings and then a lighter set of files will be served directly from the camera, not asking to retrieve external resources.

TimMahood, 2025-09-16:
not to necro this, but I had same issue,  this was down to my PiHole (DNS sinkhole) blocking some things.