themactep/thingino-firmware #457: No Audio Sources or Audio Encoders are available via ONVIF.

shamlord, 2025-01-23:
This is confirmed with the T31AL Thingino build from 19 JAN 2025.

When ONVIF is enabled in Thingino, there is no ONVIF audio profile specified by the camera. This prevents audio from being decoded or recorded with NVR software.

Using an ONVIF diagnostic tool, the following is reported from my T31AL Wyze v3 with Thingino:
[image]
Profile 0 and Profile 1 differ only in the resolution and bitrate of course.

Here is a Reolink camera with the same diagnostic tool:
[image]
There is Audio Source and Audio Encoder entries in the ONVIF data sent by the Reolink camera.

And here is that same camera with the Audio Profile groups expanded:
[image]

I do not see this issue reported by anyone else.,

gtxaspec, 2025-01-23:
@roleoroleo do you think this is something that is offered by onvif-simple-server?

shamlord, 2025-01-23:
I looked into the github onvif-simple-server.
I found /etc/onvif.conf on the camera and edited it as follows.

For Profile 0 and Profile 1, the audio parameters were originally:

> audio_decoder=AAC
> audio_encoder=NONE

I changed **NONE** to **AAC** and restarted the camera.

I now have audio via ONVIF to my NVR and I see this in that ONVIF diag tool:
[image]

Conclusion:
In /etc/onvif.conf the audio_decoder parameter is for audio being sent to the camera, and audio_encoder is for audio being sent from the camera.
With both parameters set to AAC, my NVR gives me the option to talk through the camera, though I haven't tested it yet. It may not work.

gtxaspec, 2025-01-23:
nice! we should probably set these from the webui when audio is enabled!

roleoroleo, 2025-01-23:
audio_encoder configuration is the camera's ability to encoder a signal coming from a microphone and output a stream.
audio_decoder parameter is the camera's ability to decode audio stream, so an audio back channel coming from a client and directed to the camera.
Capabilities and Profiles of the cam depend on how these parameters are set.
AudioEncoderConfiguration and AudioSourceConfiguration are enabled if audio_encoder is not NONE.
AudioDecoderConfiguration and AudioOutputConfiguration are enabled if audio_decoder is not NONE.

Often audio_encoder is not important: if the profile doesn't contain AudioSource and AudioEncoder the client might assume that the cam doesn't support audio but if the rtsp stream contains audio, probably it will work.
On the contrary, if audio_decoder is not specified, probably audio backchannel will not work; so you have to configure it in the .conf file to enable the client to send audio.
Actually, onvif_simple_server supports AAC and G711 for audio backchannel, but G711 is better in terms of compatibility.
Pay attention, the rtsp server must support backchannel, onvif is only a descriptor.

shamlord, 2025-02-04:
[onvif.conf.txt](https://github.com/user-attachments/files/18664263/onvif.conf.txt)
There does not appear to be any code currently which sets the ONVIF encoder parameters in onvif.conf to match what Streamer is actually encoding.... Here is a corrected version of the full conf file with decoder and encoder settings fixed.  I am unable to submit a suggested change to the conf file in the repo. 

Also, I suggest a check on the ONVIF webui page to ensure that onvif_notifier is advertising the correct audio codec. 
Streamer defaults to OPUS, which is not supported by ONVIF Profile S or Profile T. **ONLY** AAC and G.711 are supported, except Profile S which also supports "G726" / G.726 according to the spec.

closed 2026-03-06 by wltechblog