themactep/thingino-firmware #459: On Xiaomi MJSX03HL Motion Guard is not working - no MQTT topic brodcast, no ONVIF sensor via Home Assitant update

mackbeth11, 2025-01-23:
Hi All!

First of all appreciate your work to enable this great Xiaomi (and cheap) as real IP camera :)
I bought 3 cameras, all are this model.

My camera build is:
IMAGE_ID=xiaomi_mjsxj03hl_t31n_jxq03p
BUILD_ID="master+4ee8454, 2024-12-21 04:20:46 -0500"

I enabled in all 3 cameras motion guard and send via MQTT.
Only one camera publish it.
Settings are same in all cameras.

Working one:

[image]

MQTT settings:

[image]

and in result i can see topic via MQTT explorer:

[image]

Also on log I found that it is working:

Jan 23 10:45:26 Kamera_tyl user.info prudynt[1808]: [INFO:Motion.cpp]: Active motion detected in region 0
Jan 23 10:45:26 Kamera_tyl user.info prudynt[1808]: [INFO:Motion.cpp]: Motion Start
Jan 23 10:45:29 Kamera_tyl user.info prudynt[1808]: [INFO:Motion.cpp]: End of Motion

Other cameras which doesn;t work setup is same ,only different MQTT topic to distinguish them and debug easily.

What I found in logread for those which are not working:

Jan 23 10:28:25 Kamera_naroznik user.info prudynt[1761]: [INFO:Motion.cpp]: Start motion detection thread.
Jan 23 10:28:25 Kamera_naroznik user.info prudynt[1761]: [INFO:Motion.cpp]: Initialize motion detection.
Jan 23 10:28:25 Kamera_naroznik user.err prudynt[1761]: [ERROR:Motion.cpp]: Monitor stream is disabled, abort.

Could you help al ittle to start debugging this issue?

pabsi, 2025-01-23:
I also struggled a bit to get motion guard working (with a telegram bot in my case), but eventually it worked by rebooting the camera after enabling it.
Same for disabling it. I needed to reboot after toggling it off, otherwise it'd keep taking pictures...

mackbeth11, 2025-01-23:
I found root cause of thi issue... with disabled second stream motion is not working at all.. so it looks like motion detection is based on picture from 2nd stream..
Anyway, I try to figure out how it works, wjat is detections area, what mean sensitivity, why one of my camera detect nothing in night vision - see preview:

[image]

Also I;m interesting how to send image via MTTQ.
Currently for phone app I use Home Assistant notification, where I make snapshot -* but it takes 3 sec, so almost all of the time human is outside detection (oustide picture) when snapshot is taken :)

Is anyway good documentation what each option means?
I tried to understand whole i.e. stream options, buita rate mode, GOP... to much too detailed setting for person, whoi not di into details :) I need to setup good stream which will work with poor strenght of my WiFi outsode home..And here next question: my camera is in range of 3 WiFi, (extenders and 2 AP's) In my hoem each WiFi have SSID.. so I;m curious to which camera connects? Strongest one? randomly?
I have problems with stream stability...

pabsi, 2025-01-24:
I don't think that motion doesn't work with the second stream deactivated. Because I have them both deactivated, and motion detection works for me (on this same camera model, the Xiaomi MJSX03HL).

With regards, to your stability issues, you can try editing the `wpa_supplicant.conf` (please note I did NOT test this):

1. SSH into your camera using `ssh root@<camera_ip>`
2. Edit wpa_supplicant.conf (`nano` is not present, so you'll have to be familiar with `vi`): `vi /etc/wpa_supplicant.conf`
3. Add the `bssid` parameter to the `network` (find out what is the closes BSSID to your camera, out of the 3 you have)
4. Result (just an example):
```

root@mi2kcam-1 ~# cat /etc/wpa_supplicant.conf 
network={
	ssid="My_home_wifi"
	bssid="12:34:56:ab:cd:ef"
	psk=<64_byte_char_psk> # Assuming you have this in place, do not modify it.
}
```
5. Save the changes and reboot the camera.

I hope @themactep can get back to us on clarification regarding the motion detection.

Thank you

themactep, 2025-01-24:
Motion guard requires a stream to monitor motion activity but it can be either of two as configured.
https://github.com/gtxaspec/prudynt-t/blob/d86d8dc2e00ed066bd1b99ad4ef5a3112da6ff70/src/Motion.cpp#L133

themactep, 2025-01-24:
Check the config settings.

https://github.com/gtxaspec/prudynt-t/blob/d86d8dc2e00ed066bd1b99ad4ef5a3112da6ff70/prudynt.cfg.example#L234

mackbeth11, 2025-01-25:
Thanks a lot, i relize this setting after enabling stream 1.  Is in any place better with explanation of each setting? What features motion guard have? What mean sensitivity :1 is most insensitive where 8 is super sensitivite? I see some roi area of detection..are they option to se area of detecion? I'd like to understand, how to set detection only for large object motion like i.e. human, not get every 10 min notification  of my cat playing in backyard :) during night:) or I should use 3rd party tool like frigate?

themactep, 2025-04-04:
Closing this as resolved. Further questions are best asked in our social groups in Telegram or Discord.