themactep/thingino-firmware #461: On Dafang DF3, camera stuck or brick, don't know

Alikerks, 2025-01-24:
Hi,

I have currently a lot of Dafang DF3 patched using EliasKotlyar's procedure. Works smoothly with SS.

I thought i'll give a try to thingino on 1 spare cam but now my dafang is stuck or brick.

I tried to apply Wyze Cam Pan 1 process and i'm stuck or it is brick i can't tell.

When i start the camera, led stay blue for 5 seconds then blink quickly blue for 2 second then slowly blink blue during 30 seconds and led turn off.

Duration are approximative (~ -1/1)

Can't update demo.bin, no thingino wifi. Can't revert back to EliasKotlyar's SD nor demo.bin.

What should i do ?

Thanks

Alikerks, 2025-01-24:
resolved

Alikerks, 2025-01-24:
resolved

closed 2025-01-24 by Alikerks

QcVictor, 2025-01-25:
Hi Alikerks, just trying to achieve the same thinks, what have you done for solve this issue?

Alikerks, 2025-01-26:
> Hi Alikerks, just trying to achieve the same thinks, what have you done for solve this issue?

Hi, repeat step with autoupdate-full.bin

QcVictor, 2025-01-26:
Which method have you use for move to thingino?

Alikerks, 2025-01-26:
i use the procedure for wyze cam pan 1 but I warn you, I'm trying to revert back to dafang hack because I could not have any feed on camera local admin page nor audio through onvif nor rtsp. after trying to reinstall dafang hack, my cam got bricked and now I'm struggling to unbrick.