themactep/thingino-firmware #464: Wyze V2 requires problems

carefulcomputer, 2025-01-26:
I am using home assistant to connect my cameras (wyze v2). I noticed two issues
1. Onvif discovery in home assistant is not able to find thingino cameras. I have to manually give the IP address.
2. Thingino gives auth error when logging in from home assistant.

For 1, i do not know the solution, but for 2 , by trial and error, I found a workaround. 
The workaround  is, after first install, hard reset the camera using button at the bottom (I kept it pressed for about 15 seconds). Once it reboots camera resets and then when configuring it after that, it authenticates fine with home assistant.

csev1755, 2025-02-16:
I just tested on Home Assistant 2025.2.1 and it is also not finding my cameras. I used ODM on Windows 11 and IPCams on iOS and both of them were able to discover all of the cameras automatically. However, I was able to authenticate just fine after manually entering the IP. 

Regardless, this issue might be better suited at the source repo: [(https://github.com/roleoroleo/onvif_simple_server/issues)]

carefulcomputer, 2025-02-17:
The 'not being able to login' could be a random issue where the settings of camera get corrupted when submitting multiple changes quickly. In my testing it indicates to happen (not confirmed) when you submit a second 'save' request before first 'submit' had chance to return back.

Hypfer, 2025-03-20:
> Onvif discovery in home assistant is not able to find thingino cameras. I have to manually give the IP address.

I'm experiencing what I think is the same issue on the Wyze Pan V1

I've tracked down that in the SOAP reply to the `GetCapabilities` action, the cam replies with an IP in a bunch of `tt:XAddr` and `tt:Address` fields that is not the IP of the camera.

Instead, it's the IPv4 of the camera with the last digit cut off.

Given that it currently has an IP that results in 3 digits per octet, I am left wondering if _maybe_ somewhere some statement is off-by-one or some buffer is one byte too short. Something like that.

Hypfer, 2025-03-20:
Yep.

Assigning the camera an IP resulting in one less digit for the last octet made it work. It is now connected to Home Assistant

wltechblog, 2026-03-06:
I am assuming these issues have long sicne been fixed by newer firmware. Re-open the ticket if I'm wrong.