themactep/thingino-firmware #465: Thingino snapshot url uses admin credentials

carefulcomputer, 2025-01-26:
Thingino has two sets of passwords. One is for administration (to log into webui), and second to access the video streams.
Currently, to get the screenshot from camera, one can get it from /x/image.cgi . However that url requires administrative password and not the video stream password.
For security reasons,  admin previlieges should not be required for grabbing screenshot. It should work with streaming credentials.

themactep, 2025-01-27:
try `http://thingino:thingino@192.168.1.123/image.jpg`

carefulcomputer, 2025-01-27:
Perfect. Thanks you! I will add it to wiki.

guiguid, 2026-02-17:
Hi,
It works with  : 

http://[YOUR_CAM_IP]/x/ch0.jpg

http://[YOUR_CAM_IP]/x/ch0.mjpg