themactep/thingino-firmware #466: Issue maintaining stable connection with Unifi Protect

achecchia, 2025-01-27:
I was excited to try out Thingino and got the Wuuk PTZ as suggested by the creator. I have a unifi protect system which supports 3rd party cameras via ONVIF. when i bring the camera into Unifi Protect, it stays connected for about 8 seconds, looses connection, then reconnects, and the cycle repeats. I was wondering if it's an ONVIF standard issue or something different. i'm willing to help by submitting logs if this will help figure the issue out.

wltechblog, 2025-01-27:
Unifi Protect's ONVIF support is still new and seems to be buggy. Unfortunately it is also a closed source product so we're not in a position to lend much assistance. We have suggested to others to create support tickets with that vendor and if they are able to reach out to us we're happy to talk with them. if you have any logs with detail please by all means share them, so far their product has been a black box for us.

joeblack2k, 2025-02-02:
> I was excited to try out Thingino and got the Wuuk PTZ as suggested by the creator. I have a unifi protect system which supports 3rd party cameras via ONVIF. when i bring the camera into Unifi [...]

The current ONVIF implementation in Thingino is broken. However, there's another open-source project that sends the correct SOAP requests. You can install it on a Docker server with the camera's RTSP URLs, and it will work like magic!

https://github.com/p10tyr/rtsp-to-onvif

themactep, 2025-02-04:
@roleoroleo, take a look at this please.

roleoroleo, 2025-02-05:
> The current ONVIF implementation in Thingino is broken. However, there's another open-source project that sends the correct SOAP requests. You can install it on a Docker server with the camera's RTSP URLs, and it will work like magic!
> 
> https://github.com/p10tyr/rtsp-to-onvif

I don't know what I'm doing wrong, but I have exaclty the same behavior using this container: 10 seconds stream and 10 seconds black screen.
I installed the container and configured with this config:
```
onvif:
  - name: BulletCam
    dev: enp1s0
    target:
      hostname: 192.168.0.236
      ports:
        rtsp: 554
        snapshot: 80
    highQuality:
      rtsp: /ch0
      snapshot: /x/image.cgi
      width: 2304
      height: 1296
      framerate: 25
      bitrate: 3072
      quality: 4
    ports:
      server: 1081
      rtsp: 1554
      snapshot: 1080
```

Click to "Adopt", the stream appears and then it starts the loop stream/black screen/stream/black screen.
Test with ODM is worse: there is no stream section at all.

[image]

roleoroleo, 2025-02-05:
Tested witn ch1, same result.

gtxaspec, 2025-02-10:
merge with #490

closed 2025-02-10 by gtxaspec

signosis, 2025-05-27:
Having the same issue with Wyze Cam V2, V3, V2 Pan on a Unifi UDMPRO version 4.2.12 and protect Protect 5.3.48. Thingino verison is thingino-wyze_campan2_t31x_gc2053_atbm6031

Hatchtek, 2025-09-01:
I am having the same issue.