themactep/thingino-firmware #468: Wyze V2 camera loses network on hostname change

carefulcomputer, 2025-01-28:
In wyze v2 camera, if I change the hostname in network settings, the camera becomes unresponsive over network. Even reboot doesn't bring it back.
Only option is to hard reset using reset button at bottom and reconfigure from scratch.

themactep, 2025-02-04:
how old is the build?

carefulcomputer, 2025-02-04:
the latest. as of opening the bug. 
i have another suspicion - currently saving a change takes few seconds. if within those seconds another save request is sent,  could that possibly make the settings corrupt and requires resetting ?

themactep, 2025-02-04:
so you've set one hostname during wireless network setup, the you change it from withing web ui, right?
i need to reproduce the issue to see what's wrong.

themactep, 2025-02-04:
I flashed the latest build to a wyze v2 camera. changed hostname from web ui. rebooted the camera. no issue.

carefulcomputer, 2025-02-04:
can you please try hitting submit button while first request is still loading (in wyze v2 it takes 3-5 seconds for save to finish and http request to come back)

gtxaspec, 2025-04-01:
does this still happen on the latest build?