themactep/thingino-firmware #474: Invalid 'B' suffix in dd seek parameters

HwlloChen, 2025-02-02:
The current Makefile contains invalid usage of the 'B' suffix in dd command's seek parameters, which causes the firmware creation process to fail. 
The dd command doesn't support the 'B' suffix for byte units.

My System Environment：
OS: Ubuntu 22.04.5 LTS on Windows 10 x86_64
Kernel: 5.15.167.4-microsoft-standard-WSL2

Lines 378~391 of the current Makefile file：
```makefile
$(FIRMWARE_BIN_FULL): $(U_BOOT_BIN) $(KERNEL_BIN) $(ROOTFS_BIN) $(OVERLAY_BIN)
	$(info -------------------------------- $@)
	dd if=/dev/zero bs=$(SIZE_8M) skip=0 count=1 status=none | tr '\000' '\377' > $@
	dd if=$(U_BOOT_BIN) bs=$(U_BOOT_BIN_SIZE) seek=$(U_BOOT_OFFSET)B count=1 of=$@ conv=notrunc status=none
	dd if=$(KERNEL_BIN) bs=$(KERNEL_BIN_SIZE) seek=$(KERNEL_OFFSET)B count=1 of=$@ conv=notrunc status=none
	dd if=$(ROOTFS_BIN) bs=$(ROOTFS_BIN_SIZE) seek=$(ROOTFS_OFFSET)B count=1 of=$@ conv=notrunc status=none
	dd if=$(OVERLAY_BIN) bs=$(OVERLAY_BIN_SIZE) seek=$(OVERLAY_OFFSET)B count=1 of=$@ conv=notrunc status=none

$(FIRMWARE_BIN_NOBOOT): $(KERNEL_BIN) $(ROOTFS_BIN) $(OVERLAY_BIN)
	$(info -------------------------------- $@)
	dd if=/dev/zero bs=$(FIRMWARE_NOBOOT_SIZE) skip=0 count=1 status=none | tr '\000' '\377' > $@
	dd if=$(KERNEL_BIN) bs=$(KERNEL_BIN_SIZE) seek=0 count=1 of=$@ conv=notrunc status=none
	dd if=$(ROOTFS_BIN) bs=$(ROOTFS_BIN_SIZE) seek=$(KERNEL_PARTITION_SIZE)B count=1 of=$@ conv=notrunc status=none
	dd if=$(OVERLAY_BIN) bs=$(OVERLAY_BIN_SIZE) seek=$(OVERLAY_OFFSET_NOBOOT)B count=1 of=$@ conv=notrunc status=none
```
This code causes the following error when compiling the firmware:
```
make[1]: Leaving directory '/home/chen/thingino-n/buildroot'
-------------------------------- /home/chen/output/wyze_c2_jxf23/images/thingino-wyze_c2_jxf23.bin
dd if=/dev/zero bs=8388608 skip=0 count=1 status=none | tr '\000' '\377' > /home/chen/output/wyze_c2_jxf23/images/thingino-wyze_c2_jxf23.bin
dd if=/home/chen/output/wyze_c2_jxf23/images/u-boot-lzo-with-spl.bin bs=195516 seek=0B count=1 of=/home/chen/output/wyze_c2_jxf23/images/thingino-wyze_c2_jxf23.bin conv=notrunc status=none
dd: invalid number: ‘0B’
make: *** [Makefile:381: /home/chen/output/wyze_c2_jxf23/images/thingino-wyze_c2_jxf23.bin] Error 1
```
I used the following code to make the fix and the generated firmware passed the test on my camera: 
```makefile
$(FIRMWARE_BIN_FULL): $(U_BOOT_BIN) $(KERNEL_BIN) $(ROOTFS_BIN) $(OVERLAY_BIN)
	$(info -------------------------------- $@)
	dd if=/dev/zero bs=$(SIZE_8M) skip=0 count=1 status=none | tr '\000' '\377' > $@
	dd if=$(U_BOOT_BIN) bs=1 seek=$(U_BOOT_OFFSET) count=$(U_BOOT_BIN_SIZE) of=$@ conv=notrunc status=none
	dd if=$(KERNEL_BIN) bs=1 seek=$(KERNEL_OFFSET) count=$(KERNEL_BIN_SIZE) of=$@ conv=notrunc status=none
	dd if=$(ROOTFS_BIN) bs=1 seek=$(ROOTFS_OFFSET) count=$(ROOTFS_BIN_SIZE) of=$@ conv=notrunc status=none
	dd if=$(OVERLAY_BIN) bs=1 seek=$(OVERLAY_OFFSET) count=$(OVERLAY_BIN_SIZE) of=$@ conv=notrunc status=none

$(FIRMWARE_BIN_NOBOOT): $(KERNEL_BIN) $(ROOTFS_BIN) $(OVERLAY_BIN)
	$(info -------------------------------- $@)
	dd if=/dev/zero bs=$(FIRMWARE_NOBOOT_SIZE) skip=0 count=1 status=none | tr '\000' '\377' > $@
	dd if=$(KERNEL_BIN) bs=1 seek=0 count=$(KERNEL_BIN_SIZE) of=$@ conv=notrunc status=none
	dd if=$(ROOTFS_BIN) bs=1 seek=$(KERNEL_PARTITION_SIZE) count=$(ROOTFS_BIN_SIZE) of=$@ conv=notrunc status=none
	dd if=$(OVERLAY_BIN) bs=1 seek=$(OVERLAY_OFFSET_NOBOOT) count=$(OVERLAY_BIN_SIZE) of=$@ conv=notrunc status=none
```

gtxaspec, 2025-02-02:
could you run `dd --version` and share the output?

HwlloChen, 2025-02-02:
~/thingino-firmware (master)> dd --version
dd (coreutils) 8.32
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Paul Rubin, David MacKenzie, and Stuart Kemp.

themactep, 2025-02-02:
you need dd 9.0+. use https://github.com/themactep/docker-worker/

closed 2025-02-02 by themactep