themactep/thingino-firmware #477: webui: Green indicator at the "Services" dropdown is not visible/active anymore.

ikout, 2025-02-04:
Until a few versions back when Video Recording was enabled or Timelapse there was a "green" bar at the left of each service showing which one was active.

Since version 656853c I don't see that anymore, and I've tried 19dcece as well.
[image]

Some of the webui changes I think broke this...

themactep, 2025-02-04:
i rewrote the whole plugin/service structure and restored the indication. now it reports on the run status not just the "enabled" value. please test with tomorrow release and report back.

ikout, 2025-02-06:
Resolved with  06ac9db / 2025-02-05 release

closed 2025-02-06 by ikout