themactep/thingino-firmware #479: Winscp shell error

tranbert, 2025-02-04:
Since  thingino-aoqee_c1.bin  firmware-2025-02-03 Winscp default config cannot connect on any of my 3 cams..

"Votre shell est incompatible avec l'application (Bash est recommandé)"
"Your shell is not compatible : bash recommended"

Even if I "force" bash shell in advanced-scp/shell-shell settings...

Had to force /bin/sh to make it connect... 

Voilà...

themactep, 2025-02-04:
Like any embedded Linux, Thingino does not use Bash, only ash exposed through /bin/sh.

tranbert, 2025-02-04:
> Like any embedded Linux, Thingino does not use Bash, only ash exposed through /bin/sh.

Yes, but it was connecting before 2025-02-03 in "full auto mode" on winscp. Not a bug, but an information.

closed 2025-02-04 by tranbert