themactep/thingino-firmware #481: Firmware boot poblem on JOOAN A2R-U

Rychlous, 2025-02-05:
I uploaded firmware to the JOOAN A2R-U using a programmer. However, the camera didn’t start. I connected to the console, and I’m attaching its output. What could be the issue? Thank you.

[image]

gtxaspec, 2025-02-05:
hi, please try the latest release from today, this has been fixed.

Rychlous, 2025-02-05:
Great job, man! The camera booted up with the new firmware.

closed 2025-02-05 by Rychlous

JanezStupar, 2025-06-20:
> > [@Rychlous](https://github.com/Rychlous) what kind of a programmer did you use, if I may ask?
> 
> Hello, used cheap Chinese EEPROM/FLASH programmer - CH341.
> 
> M.

Thanks, what clip/connector did you use? Did you use the onboard port or hooked directly to the EEPROM chip? 

These might be silly questions to you, but I am quite novice in these things and based on the images of the board I have seen it is not obvious how to go about it.