themactep/thingino-firmware #487: Webhook not adding text Or image when triggered

ikout, 2025-02-06:
With the new release  06ac9db / 2025-02-05 i tried to enable webhook to see if that will get me some notifications with ntfy...
And it seems that this is working "very" partially.... :)

With "Attach Snapshot" OFF: 
It's not passing the message I add as you can see at the log it uses "-d ''" so I do get a message at ntfy but since it's empty it says "triggered"
`2025-02-06 20:45:29 [24524:webhook] curl --show-error --fail --connect-timeout 10 --max-time 30 --retry 3 --silent **-d ''** --url XXXX
`

With "Attach Snapshot" ON:
This time it uses "-F 'message='" and again not the specified message BUT also the snapshot while is send to ntfy it cannot be be opened... 
`2025-02-06 20:40:12 [22528:webhook] curl --show-error --fail --connect-timeout 10 --max-time 30 --retry 3 --silent -F 'message=' -F 'image=@/tmp/snapshot-haKEdC.jpg' --url XXXX`

`{"id":"0a4TvaX5zMOR","time":1738872028,"expires":1738915228,"event":"message","topic":"camera-hp04dHzCkq2KrY0R","message":"You received a file: attachment.bin","attachment":{"name":"attachment.bin","type":"application/octet-stream","size":684033,"expires":1738882828,"url":"https://XXXXXX/file/0a4TvaX5zMOR.bin"}}`

I'm not 100% if the attachments not working with ntfy are some compatibility issue but the "empty" message and even file names I believe should be corrected.

themactep, 2025-02-07:
you can try to use `send2ntfy`. i wrote it by a user request purely by the specs and never tested it myself as i don't use the service. the original user has vanished without even letting me know if the script works for him.

themactep, 2025-02-07:
also, i found that `send2webhook` had incorrect argument keys. fixed just now.

ikout, 2025-02-07:
Sure since I am testing ntfy myself I can try the send2ntfy, do you have an example of parameters to use with it?

themactep, 2025-02-07:
It is documented in the script itself. https://github.com/themactep/thingino-firmware/blob/master/overlay%2Flower%2Fusr%2Fsbin%2Fsend2ntfy

ikout, 2025-02-07:
Yeah, I found it and test it as well I'm not sure that this works as well.. (Similar no text/description)
If you want I can open a new issue to have that topic there.

I'll do that in a day or two, first I'll try the release that fixes the current webhook behavior, 2025-02-07 I see that has this fixed so I'll test this later today.

ikout, 2025-02-07:
With ntfy and the current webhook implementation it's not working... (correctly)

It seems that the "-F" option for curl is causing ntfy to either just say "triggered" if "Attach Snapshot" is OFF.

Or when the "Attach Snapshot" is ON then again the "Payload" message is not added to the ntfy message, and the snapshot is only read as a bin file, what i get then is this:

[image]

If I run the "send2webhook" manually without attachment it works...
`# send2webhook -v -m "Hello" -u https://XXX`

If I run it manually and use the "-s" option it "fails" and get a similar message as the screenshot.

themactep, 2025-02-07:
i pushed a new simplified version of send2ntfy. pleae give it a try. also share the command that fails for you.

ikout, 2025-02-07:
I was going to have the ntfy version on a new issue but here is the result of the new version (I copy it from github)

With this new one it seems there is a "/" at the start ?
I have ntfy locally on docker and behind reverse proxy so I use https.

```
root@Cinnado-02 ~# ./s2ntfy.sh -n "Motion Alert" -m "Cinnado 02 Motion detected"  -t https://ntfy.XXXX/camera-hp04dHzCkq2KrY0R -v
Note: Unnecessary use of -X or --request, POST is already inferred.
> POST /https://ntfy.XXXX/camera-hp04dHzCkq2KrY0R HTTP/1.1
> Host: ntfy.sh
> User-Agent: curl/8.11.0
> Accept: */*
> p:3
> Title: Motion Alert
> Content-Length: 26
> Content-Type: application/x-www-form-urlencoded
>
< HTTP/1.1 301 Moved Permanently
< Server: nginx
< Date: Fri, 07 Feb 2025 20:24:09 GMT
< Content-Length: 0
< Connection: keep-alive
< Location: /https:/ntfy.XXXX/camera-hp04dHzCkq2KrY0R
<
root@Cinnado-02 ~#
```

On the webhook topic it seems that the behavior I get is the one mentioned here:
https://docs.ntfy.sh/publish/#webhooks-publish-via-get

When there is no attachment I get the "triggered" action/message.
```
root@Cinnado-02 ~# curl https://ntfy.XXXX/camera-hp04dHzCkq2KrY0R/trigger
{"id":"4OjEmgT09EVu","time":1738960283,"expires":1739003483,"event":"message","topic":"camera-hp04dHzCkq2KrY0R","message":"triggered"}
root@Cinnado-02 ~#
```

ikout, 2025-02-07:
With the old version of send2ntfy I was getting "400 Bad Request" and I don't know if that's due to this:
> Host: ntfy.sh

I was not going to focus on the send2ntfy specifics until I was going to open the new issue just for this.

Let me know if you want to split the two topics...

themactep, 2025-02-07:
i'm fine with continuing here. it's the same service.

the command works for me.

[image]
[image]

`-t` expects just topic, not the full url. do you use a custom domain?

ikout, 2025-02-07:
Yes, as I said I self-host it with docker and have it behind a reverse proxy.
So maybe add an extra optional field for the domain?

ikout, 2025-02-07:
Can you test the snapshot upload as well? I did a quick test with the ntfy.sh and I saw this:

`Warning: You can only select one HTTP request method! You asked for both PUT (-T, --upload-file) and POST (-d, --data).`

I used "-f /tmp/snapshot-test.jpg"

themactep, 2025-02-07:
uploading an image works if there's no accompanying message. or if you send it as an attachment file (there won't be a preview in the message)

ikout, 2025-02-07:
That's strange... I'll do some more testing later/tomorrow...

ikout, 2025-02-08:
OK I think I've found a way that both attachment and message can work...

From what I've seen when you add a message you use the "-d" option that this not working correctly with the attachment since it uses the "-T" flag.

But from some testing I saw that you can pass the message as a header with something like "-H 'Message: Someone is here...'"

I did hit a small issue at the first attempt with the new "-s HOST" option that I left the last trailing "/" and that was passed at the  POST and it didn't work.

And it would be nice to add now the send2ntfy at the menus since that's only available to be run manually...

matiasdelellis, 2025-08-06:
Hi,
I also thought that webhook should be enough, but it failed. 😞 
In the end, I continued the development of themactep and added the interface [here](https://github.com/themactep/thingino-firmware/pull/715) so that ntfy works and I still have the webhook option for something else.

blackketter, 2025-09-25:
Sorry if this isn't the right place for this question, but I'm looking to try the fixed ntfy support.

I am unsure about moving from the current stable to nightly master builds.  It's not clear to me how often stable releases get pushed.

I'm currently updated to cinnado_d1_t23n_sc2336_atbm6012bx, stable+a6d16c7

Should I patch the send2ntfy script on my camera, upgrade to a nightly master build or wait for a new stable release (assuming that will be soon...)

matiasdelellis, 2025-09-25:
Hi @blackketter 
In the last few days, there has been a major rewrite across all repos... There's a master branch with this rewrite, and a stable branch that many things are missing.

The firmware images are now generated from stable, but as I said, still may be some regressions... So, I recommend you updating to a firmware image from last week prior to the changes.

I think you can pass the link to the exact firmware to install to the sysupgrade command.. 😉

themactep, 2026-08-19:
Both `send2ntfy` and `send2webhook` are fixed now, verified on hardware (Wyze Cam3 T31X) against public ntfy.sh.

## What was broken

**send2ntfy** (the proper ntfy integration):
1. **No URL scheme** — the request was always plain HTTP (`host:port/topic`), so https-only servers (e.g. `ntfy.sh:443`) answered `400 Bad Request`. The WebUI's "Use SSL" checkbox was ignored by the script.
2. **Title header was commented out** — titles never sent.
3. **Config priority ignored** — `prority=$(get_value priority)` typo meant the `priority` variable was always empty unless `-z` was passed.
4. **HTTP/2 upload stall** — curl (8.21, uclibc) uploads to ntfy.sh stalled mid-transfer at ~84% of the file (reproduced repeatedly; the request never completed). Forcing `--http1.1` fixed it: uploads complete in ~2s. Applied to the shared CURL helper so all `send2*` scripts benefit.
5. **Attachment naming** — the `Filename` header got the full `/tmp/...` path, so ntfy showed `attachment.bin` / unopenable files. Now sends the basename and only adds `-T` when the file exists → ntfy delivers a proper `image/jpeg` preview with a real filename.

**send2webhook**: the message was sent as a multipart field (`-F "message=..."`), which ntfy displayed as raw multipart garbage (`Content-Disposition: form-data; name="message"...`). Now sent as the raw POST body (`-d`) — ntfy parses it as the message; generic webhook receivers get the standard body payload. Multipart remains for actual file attachments.

## Verified on hardware (192.168.88.127, Wyze Cam3 T31X → ntfy.sh)

| Test | Result |
|---|---|
| `send2ntfy -m "Motion detected at 02:30" -n "Camera Alert"` | Message + title delivered over https |
| `send2ntfy -S` (snapshot) | JPEG attachment, proper filename, viewable — plus message+title+priority |
| `send2webhook` → ntfy | Clean message (`'webhook clean message v2'`), no multipart garbage |
| config `ntfy.priority=4` | Priority 4 delivered (typo fix) |

## Commits

- `themactep/thingino-firmware` `ciao`: **`817368f09`** ("send2: fix ntfy/webhook notifications") — package `thingino-send2` is local to this tree, so no external repo bump needed.

Upgrade to a current `ciao` build and both paths will deliver text + images properly. If you self-host ntfy with a custom domain, set host/port/Use SSL in the ntfy page (or `-h host -p port`).

closed 2026-08-19 by themactep