themactep/thingino-firmware #488: New sysupgrade Not working? Questions...

ikout, 2025-02-07:
Hello, I may be missing something so please correct me if I'm wrong....

It's my understanding that the "2025-02-05 release" was introducing a "persistent config partition"...
But I just upgraded from the 2025-02-05 to the 2025-02-07 and my config is lost again... What did I miss???

The OSD settings are reset, the Timezone config is reset, almost all things I configured after upgraded to the 2025-02-05 are lost again...
What I missed? 

Do we have docs page or list what should be "saved" after the release 2025-02-05 when doing a "sysupgrade -p"  or what I shall need to do in order to have the config "saved" correctly that future upgrades do not reset it?

gtxaspec, 2025-02-07:
the config partition is in development, and adding the partition was the first step. 
persistent configs are not working yet.

ikout, 2025-02-07:
OK, at least I'm not crazy.... 😁

Maybe add that clarification at the #484 and the [2025-02-05](https://github.com/themactep/thingino-firmware/releases/tag/firmware-2025-02-05) release notes, that this is the first step because in a way, especially the release notes make seem as the feature is ready....

closed 2025-02-07 by ikout