themactep/thingino-firmware #490: RTSP GET_PARAMETER Returns Incorrect Response Code

zzzz0317, 2025-02-08:
## Environment
- Camera Model: MJSXJ03HL
- Client Software: Unifi Protect
- Connection Protocol: ONVIF
- Streaming Protocol: RTSP

## Issue Description
When connecting to the MJSXJ03HL camera through Unifi Protect via ONVIF protocol, the video stream experiences periodic interruptions:
- Interruption frequency: Every 10 seconds
- Interruption duration: 10 seconds each time

## Technical Analysis
Through packet capture analysis, I've identified the following sequence:
1. Before each interruption, the recording software sends an RTSP GET_PARAMETER request to the camera
2. The camera responds with `RTSP/1.0 454 Session Not Found`
3. This causes the connection to drop and requires reestablishment

## Expected Behavior
According to RFC 2326 (Section 10.8: GET_PARAMETER), the camera should respond with `RTSP/1.0 200 OK` for GET_PARAMETER requests within a valid session.

## Current Behavior
The camera incorrectly responds with `RTSP/1.0 454 Session Not Found`, causing the streaming session to terminate.

## Additional Information
RFC Reference: https://www.rfc-editor.org/rfc/rfc2326.html#section-10.8

The content of the reply and response is left to the implementation. GET_PARAMETER with no entity body may be used to test client or server liveness ("ping").

## Steps to Reproduce
1. Connect MJSXJ03HL camera to network
2. Configure Unifi Protect to connect to camera via ONVIF
3. Start video stream
4. Observe connection logs or perform packet capture
5. Note the periodic interruptions and RTSP response codes

[image]

[image]

themactep, 2025-02-08:
@roleoroleo please check this out

gtxaspec, 2025-02-08:
@Lu-Fi can we force this in live555 if you recall? I'll dig in to the sources tonight.

roleoroleo, 2025-02-08:
> [@roleoroleo](https://github.com/roleoroleo) please check this out

This is exactly what I saw with my capture but I don't understand the reason.
I think this part of code is inside live555 library but I have other project based on the same library and no problem appears.
I can try to build the library with some debug info.

gtxaspec, 2025-02-08:
in:
`https://github.com/themactep/live555/blob/6d626de435a9d1e18ac247ba4ac2eba09aed4e7d/liveMedia/RTSPServer.cpp#L368`
according to this, the response is:
`  setRTSPResponse("200 OK", LIVEMEDIA_LIBRARY_VERSION_STRING);`

themactep, 2025-02-09:
Maybe httpd authentication messes things up?

roleoroleo, 2025-02-09:
A couple of updates...

Why does GET_PARAMETER receive 454 error?
When GET_PARAMETER is sent, the session is already expired and this code is executed.
```
void GenericMediaServer::ClientSession::livenessTimeoutTask(ClientSession* clientSession) {
  // If this gets called, the client session is assumed to have timed out, so delete it:

#ifdef DEBUG
  char const* streamName
    = (clientSession->fOurServerMediaSession == NULL) ? "???" : clientSession->fOurServerMediaSession->streamName();
  fprintf(stderr, "Client session (id \"%08X\", stream name \"%s\") has timed out (due to inactivity)\n",
          clientSession->fOurSessionId, streamName);
#endif
  clientSession->fLivenessCheckTask = NULL;
  delete clientSession;
}
```
So, when it tries to find the session in the lookup table it fails and `clientSession `is NULL:
```
      } else if (strcmp(cmdName, "TEARDOWN") == 0
                 || strcmp(cmdName, "PLAY") == 0
                 || strcmp(cmdName, "PAUSE") == 0
                 || strcmp(cmdName, "GET_PARAMETER") == 0
                 || strcmp(cmdName, "SET_PARAMETER") == 0) {
        if (clientSession != NULL) {
          clientSession->handleCmd_withinSession(this, cmdName, urlPreSuffix, urlSuffix, (char const*)fRequestBuffer);
        } else {
#ifdef DEBUG
          fprintf(stderr, "Calling handleCmd_sessionNotFound() (case 3)\n");
#endif
          handleCmd_sessionNotFound();
        }
```

If you add a return as a 1st line of the function `livenessTimeoutTask` it works.
But it's not the solution.

roleoroleo, 2025-02-09:
Probably I found a solution.
Please, try to increase the timeout to 20 seconds in prudynt when you create RTSPServer object:
```
...
        rtspServer = RTSPServer::createNew(*env, cfg->rtsp.port, auth, 20);
    }
    else
    {
        rtspServer = RTSPServer::createNew(*env, cfg->rtsp.port, nullptr, 20);
...
```

gtxaspec, 2025-02-10:
updated in prudynt

zzzz0317, 2025-02-11:
Thank you for fixing this issue! I've tested the new firmware version (2025-02-11) and can confirm that the RTSP connection is now stable. I ran a continuous 20-minute recording test and experienced no interruptions.

Really appreciate the quick response and resolution to this problem. 👍

closed 2025-02-11 by zzzz0317

gtxaspec, 2025-02-11:
big thanks to @roleoroleo for the detective work! thank you so much @roleoroleo !!!