themactep/thingino-firmware #491: Wireguard, and possibly other things are broken in recent firmware

ckuethe, 2025-02-08:
It apppears that the webui is inappropriately storing quoted values, or something is not stripping quotes when reading values from env. Note the `IP='"192.168.2.10/24"'` and `'x"true"'` below. I'm away from my dev machine right now - I might be able to look into this next week.

```
# sh -x /etc/init.d/S42wireguard  start
+ . /etc/init.d/rc.common
+ '[' /etc/init.d/S42wireguard '=' -sh ]
+ fw_printenv -n debug
+ debug=
+ CRONTABS=/etc/cron/crontabs/root
+ DAEMON=/etc/init.d/S42wireguard
+ DAEMON_ARGS=
+ DAEMON_PATH=/bin
+ + basename /etc/init.d/S42wireguard
sed -r 's/^[FKS]\d{2}//'
+ DAEMON_SHORT=wireguard
+ which wireguard
+ DAEMON_FULL=
+ PIDFILE=/run/wireguard.pid
+ '[' -r /etc/default/wireguard ]
+ DEV=wg0
+ fw_printenv -n wg_address
+ IP='"192.168.2.10/24"'
+ RESOLV_CONF=/tmp/resolv.conf
+ RESOLV_BAK=/tmp/resolv.conf.bak
+ starting
+ echo 'Starting wireguard'
+ log
+ logger -p daemon.info -t 'wireguard[7892]'
+ echo -n 'Starting wireguard'
Starting wireguard+ is_debug_mode
+ '[' 0 -gt 0 ]
+ '[' true '='  ]
+ echo -n ': '
: + fw_printenv -n wg_enabled
+ '[' xtrue '=' 'x"true"' ]
+ quit Disabled
+ '[' -n Disabled ]
+ log
+ echo Disabled
+ logger -p daemon.info -t 'wireguard[7892]'
+ echo_c 66 Disabled
Disabled
+ exit 0

```

gtxaspec, 2025-04-01:
is this still occurring?

pw-h, 2025-12-31:
Yes, none of the wireguard info is set to env variables when starting etc/init.d/S42wireguard at the moment, it already breaks in the first few lines  of etc/init.d/S42wireguard, appearently /etc/thingino.json is not read properly by . /usr/share/common because not a single wireguard env variable is set and /etc/init.d/S42wireguard always ends with the message "No WireGuard IP address configured".   Thingino.json is properly populated with the info in the webgui, except the public key is not read back from the file after the gui refreshes.

~# cat /etc/init.d/S42wireguard
#!/bin/sh

. /usr/share/common

if [ -z "$wireguard_address" ]; then
        echo_error "No WireGuard IP address configured"
        exit 1
fi

themactep, 2026-08-15:
Fixed — the quoted-value bug (IP='"..."', xtrue) is gone with the jct-based S42wireguard rewrite. Retest on a fresh build; if it reappears, open a new ticket.

closed 2026-08-15 by themactep