themactep/thingino-firmware #492: no preview

mat-c, 2025-02-09:
Hi

I am using 

t31x, sc4336p, 16 MB, ing-wuuk-y0510-dabe
master+5d1417a, 2025-02-08 09:15:20 +0000

After flashing preview to not work. In browser consol I see an error
```
Uncaught DOMException: An invalid or illegal string was specified
    <anonymous> http://192.168.1.19/x/preview.cgi:294
```

mpeg stream is working. timelapse is working.

In setting/streamer configuration there is lot's of entry empty (image resolution/stream disabled). It is impossible to configure it

[logread.txt](https://github.com/user-attachments/files/18723775/logread.txt)

[prudynt.cfg.txt](https://github.com/user-attachments/files/18723776/prudynt.cfg.txt)

Any idea of the issue ?

gtxaspec, 2025-02-11:
fixed in current release.

themactep, 2025-02-12:
@mat-c please confirm the fix works for you and close the issue if it does.

closed 2025-02-26 by themactep