themactep/thingino-firmware #493: ssh host key reset after reboot

mat-c, 2025-02-09:
Hi

I am using 

t31x, sc4336p, 16 MB, ing-wuuk-y0510-dabe
master+5d1417a, 2025-02-08 09:15:20 +0000

After each reboot the ssh host key is reset. This is anoying.

Host key is in /etc/dropbear that live in /var/run/dropbear

tavicu, 2025-02-09:
Are you sure only the ssh key?

I just have a power outage and after the camera booted all settings were lost.

I have set them back (onvif password, streamer settings) and then tried to reboot the camera. And again, after reboot everything was lost.

mat-c, 2025-02-09:
ssh public key for connection is ok. Wifi settings is ok.
But ssh host key is reset each time

```
Feb  9 08:44:49 ing-wuuk-y0510-dabe authpriv.info dropbear[12227]: Generated hostkey /etc/dropbear/dropbear_ecdsa_host_key, fingerprint is SHA256:xKyDM8rcZvp4cChit2uN/fm3uqVMJT8CktkMJDBQAys
```

after reboot

```
Feb  8 04:51:00 ing-wuuk-y0510-dabe daemon.info sshkeys[1139]: Restore the dropbear key (base64)
Feb  8 04:51:00 ing-wuuk-y0510-dabe daemon.info sshkeys[1139]: echo AAAAC3NzaC1lZDI1NTE5AAAAQNGYXWss67dWICRzIBOEZFJdB0735fkWwqEr/QU4UH3rj6ofJJC8g4KmXOlsux7y+uGS2WqqyOHFcl2EoP1MNA0= | base64 -d > /etc/dropbear/dropbear_ed25519_host_key
Feb  8 04:51:00 ing-wuuk-y0510-dabe daemon.info sshkeys[1139]: Read dropbear key from file
F

Feb  9 09:45:17 ing-wuuk-y0510-dabe authpriv.info dropbear[2503]: Generated hostkey /etc/dropbear/dropbear_ecdsa_host_key, fingerprint is SHA256:uD8uVqlea0URIe6t6jK8QjFZVFgL+MnlAqEmx5S/Dc8
```

I have streaming issue, so I don't know.

themactep, 2025-02-12:
How do you reboot?

themactep, 2025-03-03:
Dropbox storage directory revised in builds after March 2nd.

closed 2025-03-03 by themactep