themactep/thingino-firmware #494: exfat and wear leveling of nor flash

mat-c, 2025-02-09:
Hi new version have a config partition that is in exfat format.

This is between the u-boot env and kernel image

This is stored in exfat format that is intended to be used for sdcard (block media). Each write will update the fat and do lot's of write at the same place in flash.
For spi nor flash there won't be any wear leveling and each write will do a full erase of the block.

What are the risk to damaging the flash by doing too much write here after some time ?
Why jffs2 (or other flash fs) is not used like the overlayfs ?
If this partition is intended to have limited access, shouldn't it be mounted as ro and only remounted as rw when needed to avoid accidental damage ?

mat-c, 2025-02-09:
And if the goal is to make some files/config persistent to update, could it be possible to do : 
* keep the file in jffs2 overlay
* in update script 
** backup the file in compressed tar
** flashcp the  compressed tar in mtd config part
** trigger update
* in first boot after update (when overlay is reset)
** restore config from tar in config part

Advantage : 
* controlled and limited block write in config partition
* compression support to store more config in this tiny partition

gtxaspec, 2025-02-09:
The config partition is development in progress.  64kb is such a tiny partition, that any wear leveling  on the chip is really is moot.  We will be moving to a larger jffs2 partition as the exfat driver doesn't support whiteout files for rhe overlay.

mat-c, 2025-02-10:
On 64kb there won't be wear leveling, but lot's of erase/write cycle for each update. I think each 512B write will trigger and block/sector erase and rewrite. I don't know what is the mininal erase size.
Also with fat, there can be lot's of write in a single sector.
That's why it is recommended to run FAT over FTL.

Why don't you do like openwrt : https://openwrt.org/docs/techref/flash.layout#sysupgrade_and_rootfs_data ?

Instead of having a backup partition, they backup the file in ram. Rewrite kernel/squashfs/jffs2 and then restore in new overlay.

While waiting it to stabilize, what version is recommended to use ? The [firmware-2025-02-04](https://github.com/themactep/thingino-firmware/releases/tag/firmware-2025-02-04) before config change.

themactep, 2025-02-18:
Both rw partitions now jffs2. No exfat shenanigans. Closing as completed.

closed 2025-02-18 by themactep