themactep/thingino-firmware #500: Wyze V3 can't access after initial config

mattfox27, 2025-02-14:
Trying to setup a wyze v3 t31x with realtek chip, I can get to where I can enter wifi info and it restarts and appears to connect to wifi but I can't access it with the IP or hostname.

themactep, 2025-02-17:
@mattfox27 any progress?

mattfox27, 2025-02-17:
Nope, I get to the part where I can input my wifi info then after that nothing I try to go directly with the IP address and I try the name.local still nothing I see it pop up in my router but I can't ping it.  I have another wyze v3 with thingino installed on it and it works great so not sure what happened...I was going to try Maybe loading the stock firmware on and then redoing thingino maybe?

Nord221, 2025-02-18:
I have the Wyze V3 t31zx with realtek chipset and also encountered this same issue today when I upgraded from version master+3d0fbf8, 2025-02-07 09:18:08 +0000 to the latest 2/17/25 build.

After entering the wifi credentials on the initial config and rebooting the camera does not connect to the wifi network. Pressing the reset button just gives a chime but does not read out the IP address.

Next I long pressed the reset button to restore the camera and setup an Access Point with a predefined name. The camera rebooted but the Access Point was not named what I entered but instead says THINGINO-. I tried the password I gave it but it does not work. What worked was the password thingino. Next a dialog appeared and I entered the username root and password I gave it but it did not work. What worked was the password root.

After that I was able to connect to the UI but it is entirely bugged out and shows a white page with non clickable links. I tried long pressing to reset the camera again but the initial config no longer shows but instead a password protected THINGINO- AP appears now. It looks like the reset button does not clear out everything to default values.

Is there a way to recover from this?

FYI below is the update log that appears to be successful.

# /sbin/sysupgrade -p

Run as /sbin/sysupgrade -p
Create /tmp/sysupgrade directory
Update self

######################################################################## 100.0%

######################################################################## 100.0%
Re-run script from /tmp/sysupgrade
Run as /tmp/sysupgrade/sysupgrade -p
Check connection to https://github.com
Upgrade from GitHub
Remote file size     16777216

Remote file offset     524288
Full flash size      16777216
Available space      48234496
Download size        16252928
Download firmware from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_c3_t31x_rtl.bin

######################################################################## 100.0%
Download checksum from https://github.com/themactep/thingino-firmware/releases/latest/download/thingino-wyze_c3_t31x_rtl-update.bin.sha256sum

#=#=-#  #
-#O#- #   #
-=#=#   #   #
######################################################################## 100.0%
Verify downloaded file
thingino-wyze_c3_t31x_rtl-update.bin: OK
Clean up
removed 'thingino-wyze_c3_t31x_rtl-update.bin.sha256sum'
Relocate downloaded file
'thingino-wyze_c3_t31x_rtl-update.bin' -> '/tmp/sysupgrade/fw.bin'
Kernel detected
Stop services
Stopping onvif_notify_server: OK
Stopping vbuffer: OK
Stopping record: OK
Stopping ONVIF discovery: OK
Stopping impconfig: OK
Stopping iad: OK
Stopping prudynt: OK
Stopping rc.local: OK
Stopping telegrambot: OK
Stopping mosquitto: OK
Stopping mdnsd: OK
Stopping crond: OK
Stopping ntpd: OK
Stopping portal: Not running
Stopping sshkeys: OK
Stopping rootpassword: Same root password
Stopping rootauthkeys: No root authorized keys to back up
Stopping mdev: OK
Stopping thingino-button: OK
Stopping modules: OK
Stopping mmc: OK
Stopping ircut: OK
Stopping usb: OK
Stopping led: OK
Stopping gpio: OK
Stopping hostname: OK
Stopping sysctl: OK
Stopping dusk2dawn: OK
Stopping timezone: OK
Stopping syslogd: OK
Stopping seedrng: OK
Remove snapshot
Sync changes
Drop caches
Unmount config
'/bin/busybox' -> '/tmp/sysupgrade/busybox'
'/sbin/sysupgrade-stage2' -> '/tmp/sysupgrade/sysupgrade-stage2'
set up echo
set up flash_eraseall
set up flashcp
set up reboot
set up sh
set up sleep
Switch to Stage 2.

stage 2 PATH is /tmp/sysupgrade
Running Stage 2 of sysupgrade.
Flashing firmware to /dev/mtd6

Erasing block: 496/496 (100%)

Writing kb: 15872/15872 (100%)

Verifying kb: 15872/15872 (100%)
Rebooting in 5 seconds...
--- sysupgrade exit! ---

mattfox27, 2025-02-18:
Yes I tried to ping it in terminal and no reply

themactep, 2025-02-18:
Make full upgrade from sd card (autoupdate-full.bin). Today's release has critical breaking changes. Follow our telegram or discord to be on top of such news.

Nord221, 2025-02-18:
Thanks for your help, I did a full upgrade from the sd card and my camera is working again. I'll be sure to check discord next time to know if there are any potential breaking changes.

closed 2025-03-03 by themactep