themactep/thingino-firmware #502: Help starting up camera in flash mode

elvistony, 2025-02-14:
Just came across your awesome repo and found out I have a couple of cameras that are supported in the home page!

[image]
**Product**: Jooan A2R-U
**Flash Chip**: XMC **25QH64D**HIQ
**Other Chip**: Ingenic T23

**Host System**: Windows 11

## My attempt to booting to flash mode:

- Installed the Cloner Device drivers
- Launched the Cloner selecting the option:
[image]
- Started the (Backup mode) with the Start button
- Shorted the Pins 5,6 on the board (8 connector Flash chip) diagonally opposite to the Dot on the board
- Connected the USB cable
- _**Issue: No device found in device manager or Cloner**_

Any thoughts?

Pictures:
[image]
[image]

elvistony, 2025-02-14:
#481 : Might be the same product!

themactep, 2025-02-14:
Cloner does not work on jooan cameras. Use programmer with a clip.

closed 2025-03-01 by themactep