themactep/thingino-firmware #503: 360_ap1pa3, can't use SD card

ycsongx, 2025-02-16:
i use the bin file from git, and do some tests.
this camera can't load sd card, this is the dmesg.
----------------
[   44.343524] jzmmc_v1.2 jzmmc_v1.2.0: card inserted, state=0
[   47.532911] mmc0: error -145 whilst initialising SD card
[   49.741692] mmc0: error -145 whilst initialising SD card
[   51.951022] mmc0: error -145 whilst initialising SD card
[   81.806391] jzmmc_v1.2 jzmmc_v1.2.0: card removed, state=0
----------------
1. 
i think it may cause by the gpio settings,  so i test some other pin for gpio_mmc_cd and  gpio_mmc_power,  like 61i ....
i get these pins from other cameras , no use.
2. 
i search MMC in fireware code， and  i  get some info from my last canara D706 ,use openipc.
i think it may cause by MMC config in board.
i find that CONFIG_MMC_BLOCK is not setted in board for kernel 3.10, but setted for 4.X.
i set BR2_PACKAGE_THINGINO_KOPT_MMC=y and BR2_PACKAGE_THINGINO_KOPT_MMC0_BOOT=y   in    configs/cameras/360-pa1ap3
and build a new bin file, i use the update.bin to update carmea, still no use. 

dmesg [no sd card when boot] 
---------------------------
[    0.751676] jzmmc_v1.2 jzmmc_v1.2.0: vmmc regulator missing
[    0.757462] jzmmc_v1.2 jzmmc_v1.2.0: card-detect gpio pin not specified, card-detect disabled, enabling card_present as default.  Error: host->pdata->removal = 1, errno=-1
[    0.773598] jzmmc_v1.2 jzmmc_v1.2.0: register success!
[    0.779075] jzmmc_v1.2 jzmmc_v1.2.1: vmmc regulator missing
[    0.785091] jzmmc_v1.2 jzmmc_v1.2.1: register success!
[    0.899777] mmc0: host doesn't support card's voltages
[    0.908005] mmc0: error -22 whilst initialising SDIO card
[    0.938459] mmc0: host doesn't support card's voltages
[    0.943769] mmc0: error -22 whilst initialising MMC card
[    1.070533] mmc0: host doesn't support card's voltages
[    1.078474] mmc0: error -22 whilst initialising SDIO card
[    1.108945] mmc0: host doesn't support card's voltages
[    1.116856] mmc0: error -22 whilst initialising MMC card
[    1.250600] mmc0: host doesn't support card's voltages
[    1.255901] mmc0: error -22 whilst initialising SDIO card
[    1.289006] mmc0: host doesn't support card's voltages
[    1.299803] mmc0: error -22 whilst initialising MMC card
[    8.915265] jzmmc_v1.2 jzmmc_v1.2.1: card insert via sysfs
[    9.088657] mmc1: new SDIO card at address 0001
---------------------------

dmesg [insert sd card when boot] 
--------------------------
[    0.755659] jzmmc_v1.2 jzmmc_v1.2.0: vmmc regulator missing
[    0.761562] jzmmc_v1.2 jzmmc_v1.2.0: card-detect gpio pin not specified, card-detect disabled, enabling card_present as default.  Error: host->pdata->removal = 1, errno=-1
[    0.777608] jzmmc_v1.2 jzmmc_v1.2.0: register success!
[    0.783052] jzmmc_v1.2 jzmmc_v1.2.1: vmmc regulator missing
[    0.789111] jzmmc_v1.2 jzmmc_v1.2.1: register success!
[    2.957932] mmc0: error -145 whilst initialising SD card
[    5.157935] mmc0: error -145 whilst initialising SD card
[    7.357936] mmc0: error -145 whilst initialising SD card
[    8.865360] jzmmc_v1.2 jzmmc_v1.2.1: card insert via sysfs
[    9.038708] mmc1: new SDIO card at address 0001
-------------------------------

 [i set some config in board config file, may cause the error Line 2]

3.  i delete all files, just add  BR2_PACKAGE_THINGINO_KOPT_MMC=y  in    configs/cameras/360-pa1ap3
    update camera with the full.bin , no use
----------------------------------------
[    0.120382] jz-dwc2 jz-dwc2: cgu clk gate get error
[    9.014630] pwm-jz pwm-jz: PWM channel 2 not available (error -517), skipping...
[  111.524779] mmc0: error -145 whilst initialising SD card
[  113.733368] mmc0: error -145 whilst initialising SD card
[  115.952274] mmc0: error -145 whilst initialising SD card
----------------------------------------

gtxaspec, 2025-02-16:
I have this same camera, i will test SD card.

gtxaspec, 2025-02-16:
you are correct, I have the same errors.  I will pull this unit from production to fix the memory card, please stay tuned

ycsongx, 2025-02-16:
thanks.  i will close this issue.

by the way, i have other issues:
1.  i use tinycam-pro app, onvif needs ssl, so i try to add ssl to httpd, but it's hard for me.
     i try to use nginx, as a proxy,  it works , but no stream in webui, it is caused by sud_filter modlue in nginx,
     so i add it in nginx/objs/auto_config.......  but the nginx -V  with   sud_filter  still can't work , no stream in webui, onvif is ok.

2.  i set wifi to ap mode, and use firfox and edge login in the web, [last week] it looks like can't works well, app by onvif is ok

closed 2025-02-16 by ycsongx

gtxaspec, 2025-02-16:
run
`fw_setenv gpio_mmc_power 52o`
reboot.
this will be included in the next build of the firmware. please report if this fixes sd card.

ycsongx, 2025-02-16:
thanks ,it works,  also auto mount.

---------
[    8.397928] mmc0: new high speed SDHC card at address 0002
[    8.400644] mmcblk0: mmc0:0002 00000 3.70 GiB 
[    8.402288]  mmcblk0: p1
[   10.264851] jzmmc_v1.2 jzmmc_v1.2.1: card insert via sysfs
[   10.438685] mmc1: new SDIO card at address 0001
[   94.756792] jzmmc_v1.2 jzmmc_v1.2.0: card removed, state=0
[   94.756880] mmc0: card 0002 removed
[  109.797834] jzmmc_v1.2 jzmmc_v1.2.0: card inserted, state=0
[  111.178445] mmc0: new high speed SDHC card at address 0002
[  111.182515] mmcblk0: mmc0:0002 00000 3.70 GiB 
[  111.184766]  mmcblk0: p1
[  125.878821] jzmmc_v1.2 jzmmc_v1.2.0: card removed, state=0
[  125.878975] mmc0: card 0002 removed
[  138.739417] jzmmc_v1.2 jzmmc_v1.2.0: card inserted, state=0
[  140.019505] mmc0: new high speed SDHC card at address 0001
[  140.023361] mmcblk0: mmc0:0001 SD16G 29.5 GiB 
--------- 
/dev/mmcblk0p1            3.7G     16.0K      3.7G   0% /mnt/mmcblk0p1

/dev/mmcblk0p1           24.5G      1.2G     23.3G   5% /mnt/mmcblk0p1

i want to know how to find the gpio pin for wifi   ircut  motor ....... , i may buy other cameras for test.
i get this 360_ap1pa3 , to use the bin in git repo,  to test and learn , and i also want to make it with other camears.

gtxaspec, 2025-02-16:
please read the wiki
https://github.com/themactep/thingino-firmware/wiki
and start here:
https://github.com/themactep/thingino-firmware/wiki/Getting-Started

If you read through the wiki, you can gather and learn good information.

ycsongx, 2025-02-16:
ok, i will read all the wiki