themactep/thingino-firmware #504: Video recorder ignores storage limit

Jorgi-1, 2025-02-16:
Video recording ignores storage limit set to 15gb and continues filling up SD card until no space left and doesn't delete the oldest recordings, issue continues after reboot and manually deleting recordings

Firmware from 2025-02-12 (wyze_c2_jxf23)

`# cat /etc/webui/record.conf

record_device_path="thingino/records"
record_duration="60"
record_enabled="true"
record_filename="%Y%m%d/%Y%m%dT%H%M%S"
record_limit="15"
record_mount="/mnt/mmcblk0p1"
record_videofmt="mp4"
`

`SD card partitions
# df -h | grep 'dev/mmc'

/dev/mmcblk0p1           29.5G     29.5G         0 100% /mnt/mmcblk0p1
`

themactep, 2025-02-17:
i need logs. run `record` in console and save the output.

Jorgi-1, 2025-02-17:
> i need logs. run `record` in console and save the output.

Hi, here is the log from running record, it seems to think "thingino/records/20250212/" is the oldest but "thingino/records/20250211" is the oldest with existing recording files, upon running record it will create one recording with the latest date and time and then stop,

Thank you for the help.

[thingino.log](https://github.com/user-attachments/files/18826731/thingino.log)

themactep, 2025-02-17:
please also share the full list of files. run `ls -lR /mnt/mmcblk0p1/thingino/records/ | tb` and share the url.