themactep/thingino-firmware #505: ch1 substream not presenting codec parameters, fails in ffmpeg and others. vbuffer related?

wltechblog, 2025-02-16:
Discussing this issue on Discord but tracking it here.

substream channel isn't functioning properly in most software (mpv and go2rtc seem to work).

```[rtsp @ 0x152004080] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size
Consider increasing the value for the 'analyzeduration' (1000000) and 'probesize' (1000000) options
Input #0, rtsp, from 'rtsp://thingino:thingino@192.168.82.246:554/ch1':
  Metadata:
    title           : thingino prudynt
    comment         : stream1
  Duration: N/A, start: -0.000250, bitrate: N/A
  Stream #0:0: Video: h264, none, 90k tbr, 90k tbn
  Stream #0:1: Audio: opus, 48000 Hz, mono, fltp```

disabling vbuffer seems to be a successful workaroung.

Note from @gtxaspec 
```it's trying to create another rtsp server...hmm
[ERROR:RTSP.cpp]: Failed to create RTSP server: IPv6 bind() error (port number: 554): Address in use
for the sub stream,when it should reuse the existing one
```

workaround to disable vbuffer:
```chmod -x /etc/init.d/S96vbuffer
reboot```

dwebfr2, 2025-02-16:
after workaround with disabling vbuffer and reboot
```
chmod -x /etc/init.d/S96vbuffer
reboot
```
substream ch1 works now in VLC, before it didnt work

themactep, 2025-02-17:
Yes, that is prudynt peculiarity when one client (vbuffer) wants a stream without audio. That breaks things for other clients. Will need a fix.

mattfox27, 2025-03-01:
did this get fixed?  Having issues on latest version and ffmpeg and frigate

gtxaspec, 2025-03-01:
try the latest builds, should be fixed.

mattfox27, 2025-03-05:
wierd im still having issues with streamer not working on latest build maybe bad hardware?

gtxaspec, 2025-03-05:
streamer not working, at all?

mattfox27, 2025-03-05:
No it looks like it's working but I can't pull it up in vlc

mattfox27, 2025-03-05:
So it works for about a minute and then crashes, using latest build...I think it's hardware, I have another wyze v3 same exact setup and it works flawlessly