themactep/thingino-firmware #510: Car's blinking lights causes video recording to stutter and skip seconds

Nord221, 2025-02-18:
I have a Wyze V3 flashed with the latest Thingino build 2/17/25 pointed at my driveway. Every time a car turns on their blinkers to temporarily park in front of my driveway it causes the RTSP stream and video recording to SD card to stutter and skip 2 to 6 seconds until the car turns off their blinkers.

I tried setting the Anti-Flicker setting from 0 to 2 but it made no difference. I came from using the official Wyze RTSP beta firmware and it did not have that issue. Is there any way to resolve this issue on Thingino?

themactep, 2025-02-19:
car blinking changes a region of the screen and results in more data packed into each P-frame thus more data to process and transfer. given that it happens at night time, i would also think of a significant gain change due to changes in scene brightness which can trigger day/night mode switching.

try to identify where exactly the bottleneck is: data processing on the camera, data transfer over wifi, somewhere else. hijack your stock app to get access to the linux shell and compare the load and isp settings in similar conditions.

themactep, 2025-02-19:
run streamer in foreground (`killall prudynt; prudynt`) and check the output when lights are flashing.

one little issue with wyze rtsp beta. they forgot to opensource it. lol