themactep/thingino-firmware #513: Telegram bot no longer working on master+ad2e7ab

mpmc, 2025-02-19:
Upgraded to latest firmware using `sysupgrade -f`, telegram bot no longer works, but using `send2telegram snap` does, no commands are shown in thew webui but are present in `/etc/web.conf`.

```
NAME=Thingino
ID=thingino
VERSION="1 (Ciao)"
VERSION_ID=1
VERSION_CODENAME=ciao
PRETTY_NAME="Thingino 1 (Ciao)"
ID_LIKE=buildroot
CPE_NAME="cpe:/o:thinginoproject:thingino:1"
LOGO=thingino-logo-icon
ANSI_COLOR="1;34"
HOME_URL="https://thingino.com/"
ARCHITECTURE=mips
IMAGE_ID=aoqee_c1
BUILD_ID="master+ad2e7ab, 2025-02-18 09:38:02 +0000"
BUILD_TIME="2025-02-18 09:38:02 +0000"
COMMIT_ID="master+ad2e7ab, 2025-02-18 09:21:02 +0000"
BOOTLOADER=isvp_t23n_sfcnor
HOSTNAME=ing-aoqee-c1
TIME_STAMP=1733653000
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-gc3913cbb
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.02-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.02-git"
```

[image]

```
root@thingino-t23n ~# cat /etc/web.conf
admin_name="Thingino Admin"
admin_email=""
admin_telegram=""
admin_discord=""
telegrambot_users="[REDACTED]"
telegrambot_token="[REDACTED]"
telegrambot_command_0="start"
telegrambot_description_0="Start conversation"
telegrambot_script_0="echo \"Hello\""
telegrambot_command_1="help"
telegrambot_description_1="Request help"
telegrambot_script_1="echo \"Try https://thingino.com/\""
telegrambot_command_2="info"
telegrambot_description_2="Information about system"
telegrambot_script_2="cat /etc/os-release"
telegrambot_command_3="diag"
telegrambot_description_3="Gather diagnostic information"
telegrambot_script_3="thingino-diag"
telegrambot_command_4="snap"
telegrambot_description_4="Take a snapshot"
telegrambot_script_4="send2telegram -c \$chat_id snap"
telegrambot_command_5="clip"
telegrambot_description_5="Record a 5 second clip"
telegrambot_script_5="send2telegram -c \$chat_id clip"
telegrambot_command_6=""
telegrambot_description_6=""
telegrambot_script_6=""
telegrambot_command_7=""
telegrambot_description_7=""
telegrambot_script_7=""
telegrambot_command_8=""
telegrambot_description_8=""
telegrambot_script_8=""
telegrambot_command_9=""
telegrambot_description_9=""
telegrambot_script_9=""
webui_level="user"
webui_theme="dark"
telegram_attach_snapshot="true"
telegram_attach_video="true"
telegram_caption="%hostname, %datetime"
telegram_channel="[REDACTED]"
telegram_token="[REDACTED]"
motion_send2telegram="true"
```

mpmc, 2025-02-20:
Just to keep this issue updated.

firmware #https://github.com/themactep/thingino-firmware/commit/e22c67be534e28ffb4345e1890146fe557b2b95e loaded.

telegram bot now works but the send2telegram clip command results in "Video buffer not found".

eumesmamente, 2025-02-21:
> Just to keep this issue updated.
> 
> firmware-2025-02-20 loaded.
> 
> telegram bot now works but the send2telegram clip command results in "Video buffer not found".

I confirm this om
IMAGE_ID=cinnado_d1_t23n
BUILD_ID="master+ea51bae, 2025-02-21 08:45:31 +0000"

mpmc, 2025-03-10:
This seems to be fixed on latest master `firmware-2025-03-10`.

Can be closed unless someone else still has this issue.

closed 2025-03-10 by gtxaspec

eumesmamente, 2025-03-22:
Not fixed for me...
send2telegram -c XXXXXXX clip
  Video buffer not found

ARCHITECTURE=mips
IMAGE_ID=cinnado_d1_t23n
BUILD_ID="master+d33f1a3, 2025-03-20 08:49:36 +0000"
BUILD_TIME="2025-03-20 08:49:36 +0000"
COMMIT_ID="master+d33f1a3, 2025-03-20 00:25:59 +0000"
BOOTLOADER=isvp_t23n_sfcnor
BUILDROOT_NAME=Buildroot
BUILDROOT_VERSION=-gb4758c95
BUILDROOT_ID=buildroot
BUILDROOT_VERSION_ID=2025.02-git
BUILDROOT_PRETTY_NAME="Buildroot 2025.02-git"

mpmc, 2025-03-22:
> Not fixed for me... send2telegram -c XXXXXXX clip Video buffer not found
>
> ARCHITECTURE=mips IMAGE_ID=cinnado_d1_t23n BUILD_ID="master+d33f1a3, 2025-03-20 08:49:36 +0000" BUILD_TIME="2025-03-20 [...]

Login to your camera & start the vbuffer service `service vbuffer start`, see if that fixes it.

HTH.

eumesmamente, 2025-03-22:
> Login to your camera & start the vbuffer service `service vbuffer start`, see if that fixes it.
>
> HTH.
service vbuffer start
Service start not found

eumesmamente, 2025-03-22:
service start vbuffer
Starting vbuffer
Motion guard disabled
send2telegram -c XXXXXXX clip
  Video buffer not found

eumesmamente, 2025-03-22:
After anabled motion guard on web the preview dissapeared. Then  I restarted the stream on web and stop, start vbuffer on terminal. Then I can send clip from send2telegram .

mpmc, 2025-03-22:
> service start vbuffer Starting vbuffer Motion guard disabled send2telegram -c XXXXXXX clip Video buffer not found

Ah, sorry, I always get `service start vbuffer` and **`service vbuffer start`** wrong!