themactep/thingino-firmware #528: Wyze-V2 with latest thingino, day/nigh not working

ez2kal, 2025-02-24:
Wyze-V2 with latest thingino, day/nigh is not working as it should. everything is on defult in terms of gain 500 and 15000. as soon as i turn off the lights in the room, cam shows peach dark and IR LEDs do not turn on unless i manually select the "moon cresent" symbol on preview and then LED light turns on and i see things in black and white ( the way it should be) but in 1 minute it turns off again! i am suspecting it has something to do with the default CRON setup where it runs the command every one minute which leads to turning the LED off! and it wont turn on again untill i manually again press the moon cresent symbol  any idea how to fix this? Thanks

virmaior, 2025-02-25:
We can't really compare how the day/night works here with how wyze did it. It's not like that was a hardware function of the camera.

I've seen one of the phenomena you're talking about myself. What's happening is that the gain setting for day/night is such that the cron job is bouncing back and forth:
1. Minute 1: gain is too low in night -> goes to day
2. Minute 2: gain is too high in day -> goes to night
3. ...

My first solution was to change the values to make this less probable. dropping 500 to 250.
My second solution is this: Change the cron job to only do this check from 9am to 8pm. From 8pm to 9am, stick to night mode.

My third solution (on the infinite todo list) is to add some code to the daynight script to make it aware if it's looping and fine-tune the values when this occurs (e.g., in /tmp/ have it keep the last _n_ values of the check and if it's bouncing, then stick with night mode for the next _y_ minutes before considering switching back to day.

ez2kal, 2025-02-25:
Thanks for ideas. i ran some tests, the working conclusion was:
Switch to Day mode when gain drops below : 600
Switch to Night mode when gain raises above: 8000 

just that 1 minute running gap is kind of too long in a case when in a room you turn the lihgts on or off and instead of seeing the rsponse in a matter of seconds, it takes at least 1 minute till night vision led lights to turn on. i wish the time was set based on secconds instead of minutes which would make it more practical.

Thanks

virmaior, 2025-02-25:
https://stackoverflow.com/questions/9619362/running-a-cron-every-30-seconds

basically either run it with an offset or have it sleep 1 second at a time and check again.
no idea what the load would be for that.

ez2kal, 2025-02-25:
Off-setting is a smart idea! Thanks for sharing! I should do the math about the extral load it may create for 10 cameras!  I wish there was a more practical way implemented into the firmware! I was wondering why a basic option like on /off/automatic for night vision was not added in the first place or if the devs would be kind enough to add this. Thanks again.