themactep/thingino-firmware #534: Wyze-V2 with Thingino firmware installed, No Preview

ez2kal, 2025-02-28:
I installed thingino firmware on 5 of my wyze-v2cameras ( 5 more left) , the first 4 are all fine, the 5th one, i can connect to and the page loads but no preview and shows that broken camera image (attached). i tried, to reflash , but each time the same! everything is fine except no preview and also the GAIN on the top left corner under Thingino title, shows zero! so i have no idea what to do with this!

[image]

virmaior, 2025-02-28:
did you use the same firmware / install method for all 5?
(my first hypothesis is that the fifth camera has the other sensor and so needs a different firmware).

ez2kal, 2025-02-28:
HI,
Yes i used the same firmware as they all are Wyze-V2 versions , but i made sure , each time i reformatted the SD card and flashed a new image for every single camera. What i noticed was, when i was dong the initial configurations , i was changing the hostname to my camera names but for this specific camera the host name was vey short ( four digits), so when i reflashed it, this time i did not change the hostname and left the default name and everything went right and preview started working! so i followed the same patern for the next 5 camerras and everything went good. But i am not sure if it had anything to do with the hostname as i changed the frst four cameras ( although with longer host names) and they were all good too! so not sure what was really the culprit!

virmaior, 2025-02-28:
there are **two different models of WYZEv2**
one has a JXF22 the other a JXF23 
In WLtech's video on using the installer he points out that if you use the wrong one, it won't do video 
(just a hypothesis)

virmaior, 2025-02-28:
I see two choices on thingino.com : 
https://thingino.com/

ez2kal, 2025-02-28:
This is the only one with version 2:
https://github.com/wltechblog/thingino-installers/blob/main/wyze-cam-2/wyze-cam-2-sd.zip

I can't see the second one!

themactep, 2025-03-01:
@virmaior refers to the actual firmware while you do to the installer.

Run `thingino-diag` on the camera and post the link here.

themactep, 2025-03-05:
any progress?

closed 2025-03-08 by themactep

ez2kal, 2025-03-08:
Like I explained before, when I reflashed one last time, I tried not to change the hostname and only added password and the rest and when it was completed the review started working and so far all cameras are fine. 

Right now I am struggling with the very bad quality of streams in tinycam pro where I try to stream the cameras! I don't seem to find any option to select the stream quality! I am using automatic discovery and the cameras are very grainy!

themactep, 2025-03-08:
Did you check project wiki, article on mobie integration? It covers Tinycam Monitor app settings for thingino cameras.

ez2kal, 2025-03-08:
Any link ? No idea where it is !

themactep, 2025-03-08:
https://github.com/themactep/thingino-firmware/wiki/Integration:-Mobile-Apps

ez2kal, 2025-03-08:
I just checked the link ( thank you for sharing). My setup is exactly like that! I just used auto scan to achieve the same result! But still doesn't show any option to choose the stream quality! Looks like it is not using rtsp stream like this !

themactep, 2025-03-08:
Try to change hardware rendering to software rendering in settings of the app.

cwiggs, 2025-09-04:
I was able to fix this issue by switching to the other firmware file.  I uploaded it via the flash operations page.

[image]