themactep/thingino-firmware #540: Wyze Cam V3 - can't update firmware

boereworswizard, 2025-03-03:
Hello. I followed the non-intrusive installer guide for the Wyze Cam V3. Everything is working but the version is currently master+e22c67b, 2025-02-20

I have tried flashing the firmware and using OTA. They both say complete with no errors, however the camera becomes unresponsive, and I have to pull the power and plug it in again. Once it's running again, the camera is still on master+e22c67b, 2025-02-20

boereworswizard, 2025-03-03:
# /sbin/sysupgrade /tmp/fw-web.bin
# /sbin/sysupgrade /tmp/fw-web.bin

Run as /sbin/sysupgrade /tmp/fw-web.bin
Create /tmp/sysupgrade directory
Upgrade from a local file /tmp/sysupgrade/fw.bin
Found /tmp/fw-web.bin
'/tmp/fw-web.bin' -> '/tmp/sysupgrade/fw.bin'
Update self
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade to /tmp/sysupgrade/sysupgrade

######################################################################## 100.0%
Downloading from https://raw.githubusercontent.com/themactep/thingino-firmware/refs/heads/master/package/thingino-sysupgrade/files/sysupgrade-stage2 to /sbin/sysupgrade-stage2

######################################################################## 100.0%
Re-run script from /tmp/sysupgrade
Run as /tmp/sysupgrade/sysupgrade /tmp/fw-web.bin
Upgrade from a local file /tmp/sysupgrade/fw.bin
Found /tmp/sysupgrade/fw.bin
Upgrade from a provided file
Bootloader detected
Stop services
Stopping onvif_notify_server: OK
Stopping vbuffer: Nothing to stop
Stopping record: OK
Stopping ONVIF discovery: OK
Stopping iad: OK
Stopping prudynt: OK
Stopping rc.local: OK
Stopping telegrambot: OK
Stopping mdnsd: OK
Stopping httpd:
--- sysupgrade exit! ---

themactep, 2025-03-05:
ssh to the camera and run `sysupgrade -f`. partial upgrade won't work for your older firmware.

closed 2025-03-05 by themactep