themactep/thingino-firmware #549: S09motor: handle devices with a single set of gpios

gtxaspec, 2025-03-09:
we can either handle in the code, or assign -1 to each gpio ( this doesn't currently work )

themactep, 2025-03-11:
i'll need to check out the motor driver to see how it's handled there. ideally, we should have all peculiar settings handled in configuration to keep the driver minimal and straightforward. drivers task is to energize the pins, nothing more. for the cameras reusing the same pins for both motors driver initialization should contain all pins in the desired order for both axis plus the switching pin. this way it's easy to handle in the init script, adjust and enhance as needed without recompiling the driver.

themactep, 2025-03-13:
I rewrote the driver to use just what is sent in options, no business logic inside the driver, everything is done in configs. It works fine on jooan cameras using the same pins and a switch pin. But i observe a weird behaviour while resetting coordinates (homing). Camera rotates halfway through to one side, then to the other side, then paused in an awkward position. Manual movement with motors app works fine.

themactep, 2025-03-19:
does it work for you now?

gtxaspec, 2025-04-01:
working.
Must duplicate the single set for both H and V.

closed 2025-04-01 by gtxaspec